mne-tools/mne-python #6093: ENH: Facilitate PyTorch integration

larsoner, 2019-03-27:
From talking to @kingjr it would be nice to facilitate integration with PyTorch for backpropagation of error, for example when doing:
```
make_pipeline(SpatialFilter, ReceptiveField)
```
A good first goal would be to make an advanced example (with whatever MNE utils we need) that uses `sktorch` to get our sklearn-compatible objects to work with PyTorch with backprop. Bonus points if it actually shows there is a benefit to doing this over just using the forward methods.

larsoner, 2023-08-17:
Maybe this is a good one for @agramfort at some point :)

Or should we close as scopes etc. have changed to where this no longer makes sense?

agramfort, 2023-08-18:
this is not going happen anytime soon... closing

closed 2023-08-18 by agramfort