mne-tools/mne-python #6094: ENH: Using Freesurfer talxfm for affine alignment in SourceMorph

larsoner, 2019-03-27:
Freesurfer already provides a `talxfm` that gives the alignment of each subject to `fsaverage`/MNI space. So the step in `SourceMorph` where it computes the affine from subject A to subject B can be computed directly from these `talxfm` files rather than re-estimated.

jasmainak, 2019-03-28:
I suppose you would recompute it if the file cannot be found? Personally, I tend to prefer a little extra computation rather than intermediate files which can be missing ...

larsoner, 2019-03-28:
The way `compute_source_morph` is structured, it's assumed already that `subject_from` and `subject_to` are (in the vast majority of cases) already Freesurfer-reconstructed subjects. So for most users this requirement should be satisfied. In the meantime it makes things more consistent when someone uses a Freesurfer environment, and avoids having two ways of defining / computing the same thing, which will inevitably not give the same results. There is probably also a speed benefit from not recomputing the transformation but this is not the primary one in my mind, though it's nice to have.

larsoner, 2021-02-19:
YAGNI

closed 2021-02-19 by larsoner