mne-tools/mne-python #6096: Only 0 head digitization points of the specified kinds ("eeg", "extra",), at least 4 required

SophieHerbst, 2019-03-28:
Dear MNE developers,
we are having trouble running maxfilter on an MEG data file recorded at NeuroSpin - 
apparently we are missing some head digitization points.
We did digitize extra points and the HPI fitting went fine, so I am not sure what went wrong.
Do you have any ideas on how to debug the problem and still make maxfilter run?
Thank you!
Sophie 

>   File "/home/sh252228/anaconda3/envs/mne/lib/python3.6/site-packages/mne/bem.py", line 931, in get_fitting_dig
>     raise ValueError(msg + ', at least 4 required')
> 
> ValueError: Only 0 head digitization points of the specified kinds ("eeg", "extra",), at least 4 required

Here is the code:
`  raw_sss = mne.preprocessing.maxwell_filter(
            raw,
            calibration=config.mf_cal_fname,
            cross_talk=config.mf_ctc_fname,
            st_duration=None,
            origin='auto',
            destination=destination)`

Destination is info['dev_head_t'] from the same run.

Here is the raw.info['dig']: 

> [{'kind': 1, 'ident': 1, 'r': array([-7.1110763e-02, -1.4901161e-08, -1.4901161e-08], dtype=float32), 'coord_frame': 4}, {'kind': 1, 'ident': 2, 'r': array([ 7.4505806e-09,  9.4172612e-02, -1.4901161e-08], dtype=float32), 'coord_frame': 4}, {'kind': 1, 'ident': 3, 'r': array([ 7.0182838e-02, -1.4901161e-08, -1.4901161e-08], dtype=float32), 'coord_frame': 4}, {'kind': 2, 'ident': 1, 'r': array([0.05354978, 0.08048022, 0.05662574], dtype=float32), 'coord_frame': 4}, {'kind': 2, 'ident': 2, 'r': array([ 0.06554938, -0.02938803, -0.01712441], dtype=float32), 'coord_frame': 4}, {'kind': 2, 'ident': 3, 'r': array([-0.06564789, -0.02865517, -0.00968074], dtype=float32), 'coord_frame': 4}, {'kind': 2, 'ident': 4, 'r': array([-0.05240473,  0.08593322,  0.04769342], dtype=float32), 'coord_frame': 4}]

agramfort, 2019-03-28:
can you share the file with me?

larsoner, 2019-03-28:
And in the meantime you can get around this by manually specifying the `origin` param, e.g. `origin=(0., 0., 0.04)`.

SophieHerbst, 2019-03-28:
Thank you, both!
The file is here:
ftp://ftp.cea.fr/incoming/y2k01/herbst-sophie-1553780743/s190301_TimeInWM_run01_ERROR_raw.fif

larsoner, 2019-03-28:
If we make this a tiny bit more readable:
```
In [18]: pprint.pprint(info['dig'])
[{'coord_frame': 4,
  'ident': 1,
  'kind': 1,
  'r': array([-7.1110763e-02, -1.4901161e-08, -1.4901161e-08], dtype=float32)},
 {'coord_frame': 4,
  'ident': 2,
  'kind': 1,
  'r': array([ 7.4505806e-09,  9.4172612e-02, -1.4901161e-08], dtype=float32)},
 {'coord_frame': 4,
  'ident': 3,
  'kind': 1,
  'r': array([ 7.0182838e-02, -1.4901161e-08, -1.4901161e-08], dtype=float32)},
 {'coord_frame': 4,
  'ident': 1,
  'kind': 2,
  'r': array([0.05354978, 0.08048022, 0.05662574], dtype=float32)},
 {'coord_frame': 4,
  'ident': 2,
  'kind': 2,
  'r': array([ 0.06554938, -0.02938803, -0.01712441], dtype=float32)},
 {'coord_frame': 4,
  'ident': 3,
  'kind': 2,
  'r': array([-0.06564789, -0.02865517, -0.00968074], dtype=float32)},
 {'coord_frame': 4,
  'ident': 4,
  'kind': 2,
  'r': array([-0.05240473,  0.08593322,  0.04769342], dtype=float32)}]
```
You have two "kind"s here, the first are 1 which is fiducial, the second is 2 which is HPI. And this is confirmed with:
```
info = mne.io.read_info('s190301_TimeInWM_run01_ERROR_raw.fif')
mne.viz.plot_alignment(info, surfaces=(), dig=True, eeg=True)
```
Showing just fiducial + HPI:

[image]

If you really want to fit to these points, you can do:
```
>>> r, origin, _ = mne.bem.fit_sphere_to_headshape(info, dig_kinds=('cardinal', 'hpi'))
RuntimeWarning: Only 7 head digitization points of the specified kinds ("cardinal", "hpi",), fitting may be inaccurate
array([0.00028148, 0.01316167, 0.03819864])
```
I can make a small PR to improve the `maxwell_filter` docs on this and make a nice `__repr__` for the `dig` entries to make them easier to read. Would that have helped?

SophieHerbst, 2019-03-28:
Thank you Eric! So that means that the additional points we digitized on the head were not saved. 
To be honest, I wasn't even aware that Maxfilter relies on these additional points. I thought they would be used only for source reconstruction. 

> I can make a small PR to improve the maxwell_filter docs on this and make a nice __repr__ for the dig entries to make them easier to read. Would that have helped?

That would be great!

closed 2019-03-29 by larsoner