mne-tools/mne-python #6097: minor: peak_finder function has the same name as the module

mmagnuski, 2019-03-28:
This is just a small thing but I thought I'll report it anyway.
I remember @agramfort saying that we don't want to have functions with the same name as the name of the modules they live in. However this is the case with `peak_finder`:
```python
from mne.preprocessing.peak_finder import peak_finder
```
If one tries to import `peak_finder` directly from `preprocessing` they get a module:
```python
from mne.preprocessing import peak_finder
type(peak_finder) # module
```

agramfort, 2019-03-28:
this needs a fix indeed

>

mmagnuski, 2019-03-30:
@agramfort `peak_finder` module has only one content: the `peak_finder` function. The easiest would be to put rename the module to `_peak_finder` and `from _peak_finder import peak_finder` in the prepocessing `__init__` but that would break existing user code.

larsoner, 2019-03-31:
@mmagnuski I think we can get things working with a little bit of `sys.modules` trickery:

https://stackoverflow.com/questions/1060796/callable-modules

Feel free to take over WIP PR #6105 if you are motivated to

closed 2019-04-01 by larsoner