mne-tools/mne-python #6101: interactivity in epochs.plot_topo_image requires plt.ion()

drammock, 2019-03-29:
It appears that the "click to magnify this channel" interactivity only works if the user has already imported pyplot and called `plt.ion()`. Try this in a fresh ipython session:

```
import os
import mne
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False)
events = mne.find_events(raw, stim_channel='STI 014')
epochs = mne.Epochs(raw, events)
epochs.plot_topo_image()
# now click on any channel's imagemap, nothing happens
# now close the plot window, and...
import matplotlib.pyplot as plt
plt.show()  # only now does the enlarged image appear
```

Of course, if the last two steps aren't performed, then those orphaned figures hang around and only show up when the *next* MNE-Python plotting call is executed.  Would it work to use `matplotlib.is_interactive()` to get the ion/ioff state before the plot call, and if necessary turn it on for the duration of the plot, and then turn it back off again?

larsoner, 2019-03-29:
> Would it work to use matplotlib.is_interactive() to get the ion/ioff state before the plot call, and if necessary turn it on for the duration of the plot, and then turn it back off again?

It's probably not safe to mess with the `ion` / `ioff` state, it means that users can unexpectedly have code execute that they did not want to.

There is probably a way to get the newly spawned figures to appear. It's possible we need to call `plt.show()` after creating any new figures. It seems like something like that should work.

In a worst case, if we can't find a way to get the new ones to show up without messing with interactive state, we can emit a warning when the user clicks that the figure cannot be shown due to `ioff()`.

drammock, 2019-03-29:
right, of course. calling `plt.show()` is a cleaner solution.

drammock, 2019-03-30:
Turns out this problem is worse with `raw.plot_psd_topo()`:

```
(mnedev) [~] $ ipython
In [1]: import os 
   ...: import mne 
   ...: sample_data_folder = mne.datasets.sample.data_path() 
   ...: sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample', 
   ...:                                     'sample_audvis_raw.fif') 
   ...: raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False) 
   ...: raw.plot_psd_topo()                                                                                                           
Effective window size : 3.410 (s)
Out[1]: <Figure size 640x480 with 1 Axes>
can only concatenate list (not "NoneType") to list
Traceback (most recent call last):
  File "/opt/miniconda3/envs/mnedev/lib/python3.6/site-packages/matplotlib/cbook/__init__.py", line 215, in process
    func(*args, **kwargs)
  File "/opt/mne-python/mne/viz/topo.py", line 259, in _plot_topo_onpick
    show_func(ax, ch_idx)
  File "/opt/mne-python/mne/viz/topo.py", line 420, in _plot_timeseries
    _setup_ax_spines(ax, vline, tmin, tmax)
  File "/opt/mne-python/mne/viz/utils.py", line 2335, in _setup_ax_spines
    xticks = sorted(list(set([x for x in axes.get_xticks()] + vlines)))
TypeError: can only concatenate list (not "NoneType") to list
In [2]: # closed window at this point, then:                                                                                          
In [3]: import matplotlib.pyplot as plt                                                                                               
In [4]: plt.show()                                                                                                                    
Traceback (most recent call last):
  File "/opt/miniconda3/envs/mnedev/lib/python3.6/site-packages/matplotlib/cbook/__init__.py", line 215, in process
    func(*args, **kwargs)
  File "/opt/miniconda3/envs/mnedev/lib/python3.6/site-packages/matplotlib/backend_bases.py", line 2679, in mouse_move
    s = event.inaxes.format_coord(event.xdata, event.ydata)
  File "/opt/mne-python/mne/viz/topo.py", line 389, in <lambda>
    ax.format_coord = lambda x, y: _format_coord(x, y, labels=labels, ax=ax)
  File "/opt/mne-python/mne/viz/topo.py", line 369, in _format_coord
    nearby_data = [(data[n], labels[n], times[n]) for n in nearby]
  File "/opt/mne-python/mne/viz/topo.py", line 369, in <listcomp>
    nearby_data = [(data[n], labels[n], times[n]) for n in nearby]
TypeError: 'NoneType' object is not subscriptable
Traceback (most recent call last):
  File "/opt/miniconda3/envs/mnedev/lib/python3.6/site-packages/matplotlib/cbook/__init__.py", line 215, in process
    func(*args, **kwargs)
  File "/opt/miniconda3/envs/mnedev/lib/python3.6/site-packages/matplotlib/backend_bases.py", line 2679, in mouse_move
    s = event.inaxes.format_coord(event.xdata, event.ydata)
  File "/opt/mne-python/mne/viz/topo.py", line 389, in <lambda>
    ax.format_coord = lambda x, y: _format_coord(x, y, labels=labels, ax=ax)
  File "/opt/mne-python/mne/viz/topo.py", line 369, in _format_coord
    nearby_data = [(data[n], labels[n], times[n]) for n in nearby]
  File "/opt/mne-python/mne/viz/topo.py", line 369, in <listcomp>
    nearby_data = [(data[n], labels[n], times[n]) for n in nearby]
TypeError: 'NoneType' object is not subscriptable
Traceback (most recent call last):
  File "/opt/miniconda3/envs/mnedev/lib/python3.6/site-packages/matplotlib/cbook/__init__.py", line 215, in process
    func(*args, **kwargs)
  File "/opt/miniconda3/envs/mnedev/lib/python3.6/site-packages/matplotlib/backend_bases.py", line 2679, in mouse_move
    s = event.inaxes.format_coord(event.xdata, event.ydata)
  File "/opt/mne-python/mne/viz/topo.py", line 389, in <lambda>
    ax.format_coord = lambda x, y: _format_coord(x, y, labels=labels, ax=ax)
  File "/opt/mne-python/mne/viz/topo.py", line 369, in _format_coord
    nearby_data = [(data[n], labels[n], times[n]) for n in nearby]
  File "/opt/mne-python/mne/viz/topo.py", line 369, in <listcomp>
    nearby_data = [(data[n], labels[n], times[n]) for n in nearby]
TypeError: 'NoneType' object is not subscriptable
Traceback (most recent call last):
  File "/opt/miniconda3/envs/mnedev/lib/python3.6/site-packages/matplotlib/cbook/__init__.py", line 215, in process
    func(*args, **kwargs)
  File "/opt/miniconda3/envs/mnedev/lib/python3.6/site-packages/matplotlib/backend_bases.py", line 2679, in mouse_move
    s = event.inaxes.format_coord(event.xdata, event.ydata)
  File "/opt/mne-python/mne/viz/topo.py", line 389, in <lambda>
    ax.format_coord = lambda x, y: _format_coord(x, y, labels=labels, ax=ax)
[... 68 lines cut ...]
  File "/opt/miniconda3/envs/mnedev/lib/python3.6/site-packages/matplotlib/backend_bases.py", line 2679, in mouse_move
    s = event.inaxes.format_coord(event.xdata, event.ydata)
  File "/opt/mne-python/mne/viz/topo.py", line 389, in <lambda>
    ax.format_coord = lambda x, y: _format_coord(x, y, labels=labels, ax=ax)
  File "/opt/mne-python/mne/viz/topo.py", line 369, in _format_coord
    nearby_data = [(data[n], labels[n], times[n]) for n in nearby]
  File "/opt/mne-python/mne/viz/topo.py", line 369, in <listcomp>
    nearby_data = [(data[n], labels[n], times[n]) for n in nearby]
TypeError: 'NoneType' object is not subscriptable
Traceback (most recent call last):
  File "/opt/miniconda3/envs/mnedev/lib/python3.6/site-packages/matplotlib/cbook/__init__.py", line 215, in process
    func(*args, **kwargs)
[... cut ...]
  File "/opt/mne-python/mne/viz/topo.py", line 369, in _format_coord
    nearby_data = [(data[n], labels[n], times[n]) for n in nearby]
  File "/opt/mne-python/mne/viz/topo.py", line 369, in <listcomp>
    nearby_data = [(data[n], labels[n], times[n]) for n in nearby]
TypeError: 'NoneType' object is not subscriptable
In [5]: # this second traceback vomit was simply from mousing over the plot area,
In [6]: # and appears to be a separate issue; it happens even if plt.ion() is called 
In [7]: # before the call to raw.plot_psd_topo()
```

larsoner, 2019-03-30:
Looks like there are indeed a few bugs, can you see if the fix isn't too bad? Or let me know if you'd rather have me look

drammock, 2019-03-30:
You can assign it to me, I'll look on Monday.

On March 29, 2019 8:05:20 PM AKDT, Eric Larson <notifications@github.com> wrote:
>Looks like there are indeed a few bugs, can you see if the fix isn't
>too bad? Or let me know if you'd rather have me look

PR #6111 (2019-04-02): MRG+1: interactivity in raw.plot_psd_topo() and epochs.plot_topo_image()
closes #6101

- allows secondary figure popup when clicking in `epochs.plot_topo_image`, even if user hasn't set `plt.ion()`
- in `raw.plot_psd_topo`, avoids `TypeError: can only concatenate list (not "NoneType") to list` when clicking on a thumbnail sensor plot
- in popup figures spawned by `raw.plot_psd_topo` clicking:
    - avoids `TypeError: 'NoneType' object is not subscriptable` error on mouseover of axes area
    - fixes x axis label
commit 9b5ae510 (2019-04-02): MRG+1: interactivity in raw.plot_psd_topo() and epochs.plot_topo_image() (#6111)

* fix: interactive raw.plot_psd_topo() and epochs.plot_topo_image() closes #6101

* change x label *after* _setup_ax_spines

* fix: use correct x unit in statusbar

* ignore matplotlib warning for plt.show() calls

* FIX: re

* FIX: Better to use our show
closed 2019-04-02 by agramfort