mne-tools/mne-python #6107: BUG: "mne.read_annotations"

psychoxiatao, 2019-03-31:
#### Describe the bug
When I used the command '**mne.read_annotations'** to read .csv file. The **onset** always is 0.  However, the **duration** and **description** work well. If I change the file into .txt, it works correctly.

agramfort, 2019-03-31:
it's because the .csv reader expects onsets to be timestamps. We should be
done is raise a meaning warning / error when onsets is not timestamps in
.csv eg suggesting to use .txt

>

cbrnr, 2019-04-01:
I think a better solution would be to add an argument to indicate whether absolute or relative timestamps are present. Then we are independent of the file extension, which is a good thing because it is not clear why .csv should have absolute timestamps and .txt relative ones. Or maybe we can even auto-detect the format of the onsets?

agramfort, 2019-04-01:
but then you have one parameter that only makes sense for one file type as
the others are unambiguous
renaming a file to .txt is easy enough I would say

>

massich, 2019-04-01:
It is true that this would lead to a vector of 0s

```py
@pytest.fixture(scope='session')
def dummy_annotation_csv_notimestamp_file(tmpdir_factory):
    """Create csv file for testing."""
    content = ("onset,duration,description\n"
               "3.14, 42, AA \n"
               "6.28, 48, BB")

    fname = tmpdir_factory.mktemp('data').join('annotations_txt.csv')
    fname.write(content)
    return fname
```

we either support it or we throw an error when its not a datetime

cbrnr, 2019-04-01:
> but then you have one parameter that only makes sense for one file type as
the others are unambiguous

The file extension shouldn't be important at all, so the parameter makes sense for any file type. I don't see the deeper motivation why CSV contains timestamps and TXT doesn't.

> renaming a file to .txt is easy enough I would say

Sure it's easy, but it is still confusing.

cbrnr, 2019-04-01:
@massich alright, this is a problem. Then maybe the docs should be clearer on the distinction between the two formats and (most importantly) if something is ambiguous we should throw an error.

agramfort, 2019-04-01:
+1 to improve the doc and raise an error

PR #6184 (2019-04-23): DOC: make sure you get a warning when reading onsets which are not timestamps
closes #6107
closed 2019-04-23 by cbrnr