mne-tools/mne-python #6112: question: puzzling results in epochs.plot_topo_image

drammock, 2019-04-02:
Setup:
```
import os
import mne
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, preload=True, verbose=False)
events = mne.find_events(raw, stim_channel='STI 014')
epochs = mne.Epochs(raw, events, preload=True)
```
I expected these two options to yield the same results, but they're very different:
```
# OPTION 1
epochs.copy().pick_types(meg=False, eeg=True).plot_topo_image()
```
[image]
```
# OPTION 2
eeg_layout = mne.channels.find_layout(raw.info, ch_type='eeg')                                                            
epochs.plot_topo_image(layout=eeg_layout)
```
[image]

Is the difference here intentional, or a bug?

larsoner, 2019-04-02:
Looks like a scaling / vmiv / vmax problem in the first example

drammock, 2019-04-02:
So, not intentional I take it. I'll have a look.

jona-sassenhagen, 2019-04-02:
For the purpose of DRY, this could default to #6078 ...

drammock, 2019-04-02:
@jona-sassenhagen it looks like `_plot_masked_image` (and its proposed public version) require/create an axes object each time called, whereas `plot_topo_image_epochs` does the `_iter_topography` thing to plot each little imagmap inside one big axes.  So changing `plot_topo_image_epochs` to use `_plot_masked_image` behind the scenes might take some heavy lifting.  

(as an aside, I noticed that `_plot_masked_image` has a `picks` argument that appears to go unused?)

jona-sassenhagen, 2019-04-02:
Hm? iter_topography creates one axis per plot, right?

https://github.com/mne-tools/mne-python/blob/9b5ae5104ecdeaea13a88273c712382bf131162c/mne/viz/topo.py#L214

So this would essentially be about changing the show_func mostly, right?
Not saying it is trivial.

Either way, the same applies to https://github.com/mne-tools/mne-python/issues/4746, which I will do during the sprint.

drammock, 2019-04-02:
ah, you're right @jona-sassenhagen.  I thought it was passing `unified=True` by default, but that must have been something else I was looking at yesterday.

drammock, 2019-04-02:
@larsoner I'm convinced that in fact both of the plots above have messed up vmin/vmax. Compare the detail plots:

```
epochs.plot_image(picks='EEG 060') 
```
[image]

versus clicking on that channel after
```
epochs.copy().pick_types(meg=False, eeg=True).plot_topo_image()
```
[image]

versus clicking on that channel after 
```
epochs.plot_topo_image(layout=eeg_layout)
```
[image]

Still working out how to correct it.

jona-sassenhagen, 2019-04-02:
I remember fixing a lot of code in epochs.plot_image to get the vlims right. Maybe check there.

Or I could indeed try to hook this up to _plot_masked_image at the sprint, so it could simply share that vlim functionality.

drammock, 2019-04-03:
I'm gonna punt on this one until the sprint, I think. I put in a draft PR for what I think should be happening, but the resulting plots don't look right (too much of the signal lands near zero => images are mostly grey/white).  I also considered an approach using `np.quantile` to set vmin/vmax at the (0.01, 0.99) quantiles to pull some color into the middle range... maybe worth exploring?

mmagnuski, 2019-04-03:
I've had this problem too some time ago, but now I rarely use `plot_topo_image`.

jona-sassenhagen, 2019-04-23:
Mayyyyybe with #6186 in we can use _set_image_lims_and_scale for this.

PR #6230 (2019-04-26): FIX: proper channel-type-specific vmin/vmax in topo image plots
closes #6112 

color limits on `epochs.plot_topo_image()` are messed up. Example code:
```py
import os
import mne
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False)
# get events
events = mne.find_events(raw, stim_channel='STI 014')
event_dict = {'auditory/left': 1, 'auditory/right': 2, 'visual/left': 3,
              'visual/right': 4, [... cut]
closed 2019-04-26 by massich