mne-tools/mne-python #6119: API: simulation modul

makkostya, 2019-04-03:
I create this issue to continue general  discussion of a simulation module API, started in #5058.
We need to design the API which should:
 - easy way to define **where**  the activity is happening (labels) (see #5924)
 - **what** is the activity (waveform)
 - **when** it is happening (event info)
 - provide as an output a proper input for `simulate_raw`.

cc @sdeslauriers @ikojcic @ngayraud 

@agramfort can you label this issue as "simulation" for codding sprint pls.

larsoner, 2019-04-03:
Closing as dup of https://github.com/conda/conda/issues/8187

larsoner, 2019-04-03:
Sorry wrong link, meant https://github.com/mne-tools/mne-python/issues/5924

larsoner, 2019-04-03:
Ahh I thought #5924 was meant to cover all of these, but looking closer I guess not. Reopening

makkostya, 2019-04-09:
Here is an example of a possible simulation script:
https://gist.github.com/makkostya/4f783028bb1324dabb45388f91e2cb8f

Note: we suppose StcSimulator to be the subclass of SourceEstimate and thus it could be treated as an stc.

@ngayraud you created your own Simulator class? can you share the code pls to discuss.

ngayraud, 2019-04-09:
Will do ASAP

ngayraud, 2019-04-10:
Ok I am somewhat up to date with everything. Here is the API I created a while ago, which answers to mostly 2 and 3.
https://gist.github.com/ngayraud/22f5d0360ba34e1873ae441b850253a1

wmvanvliet, 2019-04-13:
Just wanted to leave a note that the beamformer team also wants to run simulations during the code sprint. So we're interested in the API design too!

larsoner, 2019-04-13:
Great. First we need to decide on #6037, @agramfort is not convinced it's the best approach. Once that's done, we can work on the API of the `SourceSimulator`.

wmvanvliet, 2019-04-13:
About the proposed `SourceSimulator` API, IMO it would be much simpler to no have to define all dipoles/sources at once with complicated lists/dics datastructures, but have something like:

```python
sim = SourceSimulator(fwd)
sim.add_signal(name='signal1', location=..., waveform=...)
sim.add_signal(name='signal2', location=..., waveform=...)
sim.add_signal(name='signal3', location=..., waveform=...)
 ```

larsoner, 2019-04-13:
Sounds good to me, except that it should take src not fwd in the constructor. Mapping source( time course)s to sensors is the domain of simulate_raw and should probably stay that way

larsoner, 2019-04-22:
@ngayraud et al. here is the example of a naive API I wrote to do this sort of thing:

https://gist.github.com/larsoner/8566361d3e011c9ce34c1d321f0f4ef1

Feel free to reuse the code (not the simulator) to test out the eventual API/implementation. The result will hopefully produce images like this:

[image]

[image]

megazero1316, 2019-04-24:
hello i have a question, can the simulation API be used to simulated motor imagery EEG signal (or ERD/ERS pattern ) ?

ngayraud, 2019-04-24:
Hi, I am interested in that too, and I think that it will be able to do
that, as long as you provide it with the right parameters. If you can model
the activity and provide the regions where it occurs, it should work.

ngayraud, 2019-04-29:
Now that we have #6243 merged into master, what else do we need to implement to close this issue? From what I remember, we discussed about providing template waveforms. I can provide templates for the following evoked responses: P300, N200, combination of P300+N200, and N400. For all these waveforms, two parameters can be tuned to set the amplitude and jitter of the peak. Before I do a PR, let me know if (1) you think it's a good idea and (2) if they should be on a separate .py file, maybe named waveforms.py
Users can then create their own waveforms adding as much variability as they wand, and these can be passed to the SourceSimulator using the add_data function. An example should also be implemented - I can add one in the same PR.

agramfort, 2019-04-29:
I would say priority is now to consolidate a narrative tutorial.

>

larsoner, 2019-08-21:
Narrative in #6261, closing

closed 2019-08-21 by larsoner