mne-tools/mne-python #6123: [EASY][INFRA] Twitter bot is broken since 2019-01-31

sappelhoff, 2019-04-05:
I have mentioned this multiple times and I agree that it's not urgent at all, but I really liked our Twitter Bot "[mne_python](https://twitter.com/mne_python)" and would like to see it restored.

Perhaps this can be done at the sprint or so. Anyhow, this issue is just a reminder for us.

Previous investigation by @agramfort:

> github says: Note: GitHub Services have been deprecated. Please contact your integrator for more information on how to migrate or replace a service with webhooks or GitHub Apps.

agramfort, 2019-04-07:
if you know any solution please let us know

>

sappelhoff, 2019-04-08:
What solution did we previously use? Was it this one: https://github.com/github/github-services/blob/master/docs/twitter ?

There are several solutions, but most of them seem to involve signing up with some external service. I think that would be fine, because we could use the free plan. Some examples:

- [automate.io](https://automate.io/integration/github/twitter)
- [zapier](https://zapier.com/apps/github/integrations/twitter/1352/share-new-github-commits-on-twitter)

agramfort, 2019-04-08:
I  arrived to the same conclusion

>

sappelhoff, 2019-04-08:
@agramfort it'd probably have to be you to set it up (or someone else with credentials both for mne-tools and `@mne_python` twitter).

I suggest to go with Zapier, because their free plan seems to be a bit more permissive:

> That limit is approximately 200 attempts every 10 minutes window per zap.

whereas with Automate.io states:

> Rate limited to 1 every 5 minutes.

agramfort, 2019-04-09:
let's look at this during the sprint

>

sappelhoff, 2019-04-24:
@agramfort let me know when you have 10 minutes for this.

sappelhoff, 2019-04-24:
I investigated a bit and apparently, FieldTrip is using a whole suite of scripts to automate this themselves, without relying on 3rd party services:

https://github.com/fieldtrip/webhook

I also asked Robert whether he can tell us some details - I'll update once I know more.

sappelhoff, 2019-04-26:
update: Alex tried `zapier`, but ran into problems when selection the mne-tools ORG as the owner of mne-python (he could only select his own fork) 

We need somebody to test `automate.io` or `zapier` and document what options / clicks are needed.

sappelhoff, 2019-04-28:
another option: https://gitweet.io/

> Gitweet is a convenient tool for developers and makers that wants to quickly tweet something about their last code update.
Once your repository is linked to a twitter account, every time you provide a description in a commit (completely optional obviously), a tweet will be posted with the provided content.
Only the description will be tweeted.

sappelhoff, 2019-05-04:
@agramfort - do you remember that you tried ZAPIER during the Coding Sprint? --> It did not work, because you could not select the  "mne-tools" organization.

The solution is [here](https://zapier.com/help/common-problems-github/#why-are-my-organizations-repositories-missing). We would have needed to "grant 3rd party access" to Zapier for the organization.

Perhaps you can give it another try :-)

larsoner, 2019-05-04:
Both providers all had kind of crazy permission things (read and write all public and private repos) that would create security risks, so I created a dummy `larsoner-mne` user and put it in a dummy `mne-twitter` team on GitHub with only read access to MNE-Python, authorized it to connect to GitHub and Twitter, and have it so it should push tweets to `mne_news` like:

> Committed MRG, DOC: categorize tutorials (#6264)
> * move all tutorials and add tutorial section READMEs
> * fix crossrefs [skip travis][skip appveyor][skip azp] by Daniel McCloy (https://api.github.com/users/drammock)

Pushed the test tweet:

https://twitter.com/mne_news/status/1124651146872856576

Okay? If so we just need to see if a commit to master actually submits it, too.

@agramfort if this is the wrong twitter account, send me privately the name and p/w for the one I should use.

dengemann, 2019-05-04:
I am -1000 for using mne_news which is our human operated news accoubt. Frankly, many people do not want to see commit notifications. Can we create a separate account for that?

sappelhoff, 2019-05-04:
thanks for the initiative @larsoner and the neat solution with @larsoner-mne. Some questions for me to understand:

> put it in a dummy mne-twitter team on GitHub with only read access to MNE-Python

Do you mean you created a new "secret" team within the mne-tools organization? Because I can't see any  `mne-twitter` here: https://github.com/orgs/mne-tools/teams

> authorized it to connect to GitHub and Twitter

With "it" you mean [zapier](https://zapier.com)?

---

> Can we create a separate account for that?

@dengemann --> yes, we have [@mne_python](https://twitter.com/mne_python) for that. Alex has the login data

larsoner, 2019-05-04:
> Do you mean you created a new "secret" team within the mne-tools organization?

Yes I made it private because nobody should really need to care about its existence or what it's for

> With "it" you mean zapier?

Yes

> Can we create a separate account for that?

> @dengemann --> yes, we have @mne_python for that. Alex has the login data

Done

closed 2019-05-04 by larsoner

massich, 2019-05-14:
@larsoner if the commit message when merging a PR is tweeted, it means that we need to curate those messages so that we don't tweet something like [this](https://github.com/mne-tools/mne-python/commit/8ce77a0ad91844b0bb2a5f99bf383f1cc9d39fe6):
```
[MRG] small ICA bug fixes for #5967 - 2nd try (#6246)

* Fix sorting in descending order in ica._detect_artifacts (was sortin in ascending order).

* Fix sfreq selection in ica._band_pass_filter (called (only) in ica.score_sources to filter sources and targets) to use the raw sfreq rather than the one used for fitting the ica.

* Incorporate missed suggestion from #5620 + summary of discussion

* fix PEP8

* fix typo in doc

* Allow both exclude and ica.exclude to be array_like by always converting to list first.

* ica.detect_artifacts: Corrected doc according to code and changed some default values to yield the same compos as before

* Added test for checking descending sorting in detect_artifacts for integer criterion

* Stylistic improvements thanks to jona-sassenhagen

* Fix test for score-sorting

* pep8 and whats_new.rst

* Remove links to privat funcs in whats_new.rst

* fix links in whats_new

* STY: No end period

* Futher improvements thanks to mmagnuski and a few further corrections in the docstrings of run_ica and detect_artifacts

* Added warning for new sorting-direction in detect_artifacts; added whats_new from #6247

* improve bug section; remove warning again; typo
```

If that's the case (which I'm +1000), we should add this in our contributing guide in the section of how to review (cc: @drammock)

larsoner, 2019-05-14:
This is how it looks currently:

https://twitter.com/mne_python?lang=en

If someone wants to look at the Zapier syntax I'm happy to update the automation

massich, 2019-05-14:
I thought that clean messages would appear nice. This is an example of clean-up message https://github.com/mne-tools/mne-python/commit/b2de2e9a192301e4dfb106cef3963d1283d74024 but the tweet does not look nice either https://twitter.com/mne_python/status/1084953412326825984

I'm not sure if its worth the effort.

agramfort, 2019-05-14:
i think it's good enough

>

larsoner, 2019-05-15:
We do currently get an error for long messages:
```

MNE-Python commit -> mne_python Tweet
  |   | had 1 error (edit this Zap)
Error Message:Twitter: Tweet needs to be a bit shorter.
```

I'll see if there is a simple way to truncate on Zapier.

larsoner, 2019-05-15:
I took a minute to tell Zapier to take just the first line of the commit message. Now looks like this:

https://twitter.com/mne_python/status/1128726165487783936

larsoner, 2019-05-20:
Unfortunately for Zapier, having a 3-step requires an account, so I'm going to remove the line restriction. Those tweets can just fail.