mne-tools/mne-python #6125: The case of overlapping/simultaneous events

nbara, 2019-04-05:
Hi all,

I'd like to store (.fif) files with overlapping events. 

Currently this is not possible : 

```Python
import mne

>>> info = mne.create_info(['a', 'b'], 1., ['eeg', 'stim'])
>>> raw = mne.io.RawArray(np.random.randn(2, 100), info=info)
>>> events = np.array([[1, 0, 1],
                       [1, 0, 2],
                       [10, 0, 3],
                       [10, 0, 4]])
>>> raw.add_events(events, replace=True)
>>> events_out = mne.find_events(raw, consecutive=True)
2 events found
Event IDs: [2 4]
>>> print(events_out)
[[ 1  0  2]
 [10  0  4]]
```

Is this something that should work or am I doomed to fail ? Or maybe it's a good use case for Annotations (never used them) ?

Looking at the code, it seems that [`raw.add_events()`](https://github.com/mne-tools/mne-python/blob/56909cdbf35511d9837a5999ce1ffe2f47eee7bb/mne/io/base.py#L2125) inserts the event values into `raw._data[stim_channel]`. But I thought stim_channels were (going to be?) deprecated (cf https://github.com/mne-tools/mne-python/issues/5020) ?

larsoner, 2019-04-05:
> what about raw.info['events']

This field is not really meant to be used so far. It's something that Elekta provides. You can try to hack it to do what you want, but support for it might break at some point if we realize it has some limitations we haven't imposed yet.

So far the way to solve this problem is to use hierachical event IDs, for example if event 1 is auditory and 2 is visual, then you give this some other number (12?) and do `event_id={'auditory': 1, 'visual': 2, 'auditory/visual': 12}`. Later if you do `epochs['auditory']` it will get both `'auditory'` and `'auditory/visual'`.

It's not the most convenient in all cases, but maybe it will work for you here? We can think about if there is some way to allow duplicate events. But it will almost certainly involve some inefficiencies, such as actually storing duplicate information, and might end up breaking other things / being too complicated to be worth the effort.

nbara, 2019-04-05:
Yeah but event_ids only relate to epochs, right ? 

I have a whole bunch of numpy arrays containing raw EEG data that need to be saved as `-raw.fif` (or any MNE-compatible format) files. ATM the only solution I've found is to save events in a separate CSV file, which is really inconvenient...

larsoner, 2019-04-05:
Annotations do allow for simultaneous things. You can use those for now. They get saved with the raw data

nbara, 2019-04-05:
OK I'll look into it thanks!

Two suggestions that come to mind: 
- events can be written to file with `mne.write_events()`; it would be cool if `mne.io.read_raw_fif()` accepted an `event_file` parameter so that events would attached to the raw instance straight away (especially if there is no more stim_channel in `raw._data`
- it would be usefult if `Annotations` accepted sample numbers in combination / instead of time_stamps

larsoner, 2019-04-05:
> events can be written to file with mne.write_events(); it would be cool if mne.io.read_raw_fif() accepted an event_file parameter so that events would attached to the raw instance straight away (especially if there is no more stim_channel in raw._data

I'd rather provide a convenience function to convert a events-> annotations (probably needs the raw instance to deal with first_samp and sfreq), then attach annotations the standard way (`raw.annotations.append` or `raw.set_annotations`)

> it would be usefult if Annotations accepted sample numbers in combination / instead of time_stamps

The convenience function above would take care of this sort of conversion. The annoying parts are that for sample numbers you need to know `sfreq` (and often, if the numbers come from events, `first_samp`), which are often related to `raw`. But `Annotations` themselves are standalone objects (they can be written and read independently) and have no notion of `sfreq`, and `first_samp` is only implicitly in them somewhere in that they have an `orig_time` that relates back to a raw instance's `meas_date` and `first_samp`.

nbara, 2019-04-08:
>I'd rather provide a convenience function to convert a events-> annotations (probably needs the raw instance to deal with first_samp and sfreq), then attach annotations the standard way (raw.annotations.append or raw.set_annotations)

Do you see this as a class method or simply a utility function ?

I like `Annotations().from_events(events, raw)` as it's straightforward in terms of API.

agramfort, 2019-04-08:
we could have an

mne.annotations_from_events
to match the new
mne.events_from_annotations

>

nbara, 2019-04-08:
I like this. I can make a PR this week.

agramfort, 2019-12-28:
@nbara still willing to look into this?

nbara, 2019-12-30:
Sure, I was just waiting for confirmation from you. I'll put up a PR next week.

nbara, 2020-01-15:
closed via #7171

closed 2020-01-15 by nbara