mne-tools/mne-python #6127: bem meshes generated by brainsuite cannot be downsampled (mne.make_bem_model))

yvonnio, 2019-04-06:
Hi guys,
I'm having some issues with the BEM meshes for my EEG data.  I've tried two approaches:
(1) Using freesurfer watershed and trying different preflood levels, but it seems to do a bad job with the outer_skull.surf being misplaced (it extends down to the neck, mostly following the outer_skin.surf).  Will this be problematic for source localization?  Is there a way to overcome this?

[image]

(2) I've had better success with using Brainsuite and have followed the guidelines (https://martinos.org/mne/stable/manual/appendix/bem_model.html); converting it from dfs to surf using mne_convert_surface, and then reducing the number of triangles to 10,000 using mne_reduce_surface.  When I try to run `mne.make_bem_model`, I get the error:

> File "/usr/anaconda3/envs/mne/lib/python3.6/site-packages/mne/bem.py", line 344, in _ico_downsample
> raise RuntimeError(bad_msg)
> RuntimeError: A surface with 10000 triangles cannot be isomorphic with a subdivided icosahedron.

I can overcome this issue by setting 'ico=None' but am not sure if this affects source localization.  I've attached the .surf I generated using Brainsuite on the MNE sample dataset.
[inner_skull.surf.gz](https://github.com/mne-tools/mne-python/files/3050758/inner_skull.surf.gz)
[outer_skin.surf.gz](https://github.com/mne-tools/mne-python/files/3050759/outer_skin.surf.gz)
[outer_skull.surf.gz](https://github.com/mne-tools/mne-python/files/3050760/outer_skull.surf.gz)

larsoner, 2019-04-08:
I cannot reproduce the issue you reported on the listserv with the `sample` surfaces. I put the unzipped surfaces in `foo/bem` and ran:
```
import numpy as np
import mne
bem = mne.make_bem_model('foo', None, subjects_dir='.', verbose=True)
with np.errstate(invalid='raise', divide='raise'):
    sol = mne.make_bem_solution(bem, verbose=True)
```
I just noticed that you uploaded the `sample` dateset surfaces. If we are going to debug the error you got (with the divide by zero), we'll need the actual problematic surfaces.

larsoner, 2019-04-08:
... I see you've now edited it include the surfaces (GitHub just updated the page for me -- we are not notified of edits.) I'll take a look now.

larsoner, 2019-04-08:
Your surface has topological defects, two of the vertices have fewer than 3 neighbors. This should be fixed somehow (not sure how). My guess is that it's in the eyes based on the output of:
```
mne.viz.plot_alignment(surfaces=('inner_skull', 'outer_skull', 'outer_skin'),
                       subjects_dir='.', subject='foo')
```
See #6133 where we raise an informative error message, in your case this is:

> Surface outer skin  has topological defects: 2 / 10000 vertices have fewer than three neighboring triangles [1513, 2566]

yvonnio, 2019-04-08:
Ok, thanks for the help.  I'll keep an eye out on the pull request.