mne-tools/mne-python #6128: report does not work for toy example

jasmainak, 2019-04-06:
If I do this on master:

```py
from mne import Report
import matplotlib.pyplot as plt
rep = Report()
fig = plt.plot([1, 2], [1, 2])
rep.add_figs_to_section(fig, 'my caption')
```
it fails with the following message:

```py
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-5-a4d487ddeb86> in <module>()
      3 rep = Report()
      4 fig = plt.plot([1, 2], [1, 2])
----> 5 rep.add_figs_to_section(fig, 'my caption')

~/Desktop/projects/github_repos/mne-python/mne/report.py in add_figs_to_section(self, figs, captions, section, scale, image_format, comments, replace)
   1078             img_klass = self._sectionvars[section]
   1079 
-> 1080             img = _fig_to_img(fig, image_format, scale)
   1081             html = image_template.substitute(img=img, id=global_id,
   1082                                              div_klass=div_klass,

~/Desktop/projects/github_repos/mne-python/mne/report.py in _fig_to_img(fig, image_format, scale, **kwargs)
     74                  '`mayavi.core.api.Scene` figure instances'
     75                  ' will throw an error.' % (e,))
---> 76         if fig.scene is not None:
     77             img = mlab.screenshot(figure=fig)
     78         else:  # Testing mode

AttributeError: 'Line2D' object has no attribute 'scene'
```

agramfort, 2019-04-07:
oups @GuillaumeFavelier we broke something :(

can you have a look? or maybe @jasmainak you can find the fix quickly?

jasmainak, 2019-04-08:
sorry it's a false alarm, plt.plot() does not return a figure object. 

Nothing to worry :-)

larsoner, 2019-04-08:
It should raise an error if not a mpl or mayavi figure

PR #6132 (2019-04-11): MRG: Better type check for report
Closes #6128.
closed 2019-04-11 by agramfort