mne-tools/mne-python #6137: Motor imagery decoding from EEG data using the Common Spatial Pattern (CSP)

pengweimin, 2019-04-09:
when i run this code ，some error was appear.

  File "<ipython-input-8-096867868191>", line 1, in <module>
    runfile('C:/Users/Administrator/Desktop/deep/preprocessing/mne-python-master/plot_decoding_csp_eeg.py', wdir='C:/Users/Administrator/Desktop/deep/preprocessing/mne-python-master')

  File "C:\Program Files\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 866, in runfile
    execfile(filename, namespace)

  File "C:\Program Files\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/Administrator/Desktop/deep/preprocessing/mne-python-master/plot_decoding_csp_eeg.py", line 38, in <module>
    events = find_events(raw, shortest_event=0, stim_channel='STI 014')

  File "<C:\Program Files\Anaconda3\lib\site-packages\mne\externals\decorator.py:decorator-gen-129>", line 2, in find_events

  File "C:\Program Files\Anaconda3\lib\site-packages\mne\utils\_logging.py", line 89, in wrapper
    return function(*args, **kwargs)

  File "C:\Program Files\Anaconda3\lib\site-packages\mne\event.py", line 685, in find_events
    raise ValueError('No stim channel found to extract event triggers.')

ValueError: No stim channel found to extract event triggers.

pengweimin, 2019-04-09:
code:

import numpy as np
import matplotlib.pyplot as plt

from sklearn.pipeline import Pipeline
from sklearn.discriminant_analysis import LinearDiscriminantAnalysis
from sklearn.model_selection import ShuffleSplit, cross_val_score

from mne import Epochs, pick_types, find_events
from mne.channels import read_layout
from mne.io import concatenate_raws, read_raw_edf
from mne.datasets import eegbci
from mne.decoding import CSP

print(__doc__)

# #############################################################################
# # Set parameters and read data

# avoid classification of evoked responses by using epochs that start 1s after
# cue onset.
tmin, tmax = -1., 4.
event_id = dict(hands=2, feet=3)
subject = 1
runs = [6, 10, 14]  # motor imagery: hands vs feet

raw_fnames = eegbci.load_data(subject, runs)
raw_files = [read_raw_edf(f, preload=True, stim_channel='auto') for f in
             raw_fnames]
raw = concatenate_raws(raw_files)

# strip channel names of "." characters
raw.rename_channels(lambda x: x.strip('.'))

# Apply band-pass filter
raw.filter(7., 30., fir_design='firwin', skip_by_annotation='edge')

events = find_events(raw, shortest_event=0, stim_channel='STI 014')

picks = pick_types(raw.info, meg=False, eeg=True, stim=False, eog=False,
                   exclude='bads')

# Read epochs (train will be done only between 1 and 2s)
# Testing will be done with a running classifier
epochs = Epochs(raw, events, event_id, tmin, tmax, proj=True, picks=picks,
                baseline=None, preload=True)
epochs_train = epochs.copy().crop(tmin=1., tmax=2.)
labels = epochs.events[:, -1] - 2

###############################################################################
# Classification with linear discrimant analysis

# Define a monte-carlo cross-validation generator (reduce variance):
scores = []
epochs_data = epochs.get_data()
epochs_data_train = epochs_train.get_data()
cv = ShuffleSplit(10, test_size=0.2, random_state=42)
cv_split = cv.split(epochs_data_train)

# Assemble a classifier
lda = LinearDiscriminantAnalysis()
csp = CSP(n_components=4, reg=None, log=True, norm_trace=False)

# Use scikit-learn Pipeline with cross_val_score function
clf = Pipeline([('CSP', csp), ('LDA', lda)])
scores = cross_val_score(clf, epochs_data_train, labels, cv=cv, n_jobs=1)

# Printing the results
class_balance = np.mean(labels == labels[0])
class_balance = max(class_balance, 1. - class_balance)
print("Classification accuracy: %f / Chance level: %f" % (np.mean(scores),
                                                          class_balance))

# plot CSP patterns estimated on full data for visualization
csp.fit_transform(epochs_data, labels)

layout = read_layout('EEG1005')
csp.plot_patterns(epochs.info, layout=layout, ch_type='eeg',
                  units='Patterns (AU)', size=1.5)

###############################################################################
# Look at performance over time

sfreq = raw.info['sfreq']
w_length = int(sfreq * 0.5)   # running classifier: window length
w_step = int(sfreq * 0.1)  # running classifier: window step size
w_start = np.arange(0, epochs_data.shape[2] - w_length, w_step)

scores_windows = []

for train_idx, test_idx in cv_split:
    y_train, y_test = labels[train_idx], labels[test_idx]

    X_train = csp.fit_transform(epochs_data_train[train_idx], y_train)
    X_test = csp.transform(epochs_data_train[test_idx])

    # fit classifier
    lda.fit(X_train, y_train)

    # running classifier: test classifier on sliding window
    score_this_window = []
    for n in w_start:
        X_test = csp.transform(epochs_data[test_idx][:, :, n:(n + w_length)])
        score_this_window.append(lda.score(X_test, y_test))
    scores_windows.append(score_this_window)

# Plot scores over time
w_times = (w_start + w_length / 2.) / sfreq + epochs.tmin

plt.figure()
plt.plot(w_times, np.mean(scores_windows, 0), label='Score')
plt.axvline(0, linestyle='--', color='k', label='Onset')
plt.axhline(0.5, linestyle='-', color='k', label='Chance')
plt.xlabel('time (s)')
plt.ylabel('classification accuracy')
plt.title('Classification score over time')
plt.legend(loc='lower right')
plt.show()

agramfort, 2019-04-09:
@pengweimin what version of mne are you using? FYI it should work as our continuous integration system test this everyday and it seems to work if example version matches your mne version.

pengweimin, 2019-04-10:
> @pengweimin what version of mne are you using? FYI it should work as our continuous integration system test this everyday and it seems to work if example version matches your mne version.

mne version  0.17.2 
python 3.5.2

pengweimin, 2019-04-10:
> @pengweimin what version of mne are you using? FYI it should work as our continuous integration system test this everyday and it seems to work if example version matches your mne version.

mne v0.18.dev0(0.17.2)
python 3.5.2
thanks

cbrnr, 2019-04-10:
If you are referring to the included example script `mne-python/examples/decoding/plot_decoding_csp_eeg.py`, this works with the current master. As @agramfort mentioned this is tested every time someone makes a change to the code. I also tested it locally and it works.

Most likely you are using an old version. Please either install the latest stable version 0.17.2 or pull the current master if you need the dev version. The line which throws the error in your script (`events = find_events(raw, shortest_event=0, stim_channel='STI 014')`) does not exist in the new version anymore.

closed 2019-04-22 by agramfort