mne-tools/mne-python #6138: plot_alignment not showing EEG electrode locations on scalp

corscheid, 2019-04-09:
We're trying to verify all of our alignments. (EEG electrodes, source space, etc.)

Following [this tutorial](https://martinos.org/mne/stable/auto_examples/visualization/plot_eeg_on_scalp.html), we have created a test script derived from the example in which the only modification was to change the `*_raw.fif` data file to be loaded into the `raw` object.

The example script runs as expected, however upon replacing the raw data file with our own data file, the EEG electrodes do not show up, nor does the 3D head model. `raw.plot_sensors()` however does show the expected arrangement of electrodes (in a 2D montage plot)

No error messages are printed.

Expected (although technically with fewer channels than this):
[image]

Actual:
[image]

raw.plot_sensors():
[image]

I have spent a lot of time debugging this so far. Any ideas what's going wrong?

agramfort, 2019-04-10:
how did you specify the eeg channel locations for your raw object?
DigMontage? Montage? do you know if which coord system these positions are?

>

corscheid, 2019-04-10:
Hi agramfort, Thanks for your reply.

Here's the process for how this raw object came to be:

```python
# In a separate and previously run script:
raw = read_raw_eeglab(raw_fname, montage=montage_fname, eog=('HEOG', 'VEOG'), preload=True, verbose=True)
# ...
raw.save(fif_fname, overwrite=True)
```
```python
# in the current script which just aims to verify alignments
raw = mne.io.read_raw_fif(fif_fname)
```
where
- `raw_fname` is a .set file previously saved/exported from EEGLAB
- `montage_fname` is the `standard-10-5-cap385.elp` file copied from EEGLAB
- `fif_fname` in both cases is the same file, exported from one mne script and imported in the current one.

agramfort, 2019-04-10:
such a montage is defined on a sphere and not in the MRI space of your
subject.

do you actually have an MRI for your subject? or you want to work with a
template brain?

yvonnio, 2019-04-10:
I have a similar issue, `mne.plot_alignment` basically gives me an empty display.  I've narrowed my issue down to the an issue with a misalignment of the default dig montage and the MRI space of my subject.  Like corscheid, I'm also using a default montage but in my case, `montage  = mne.channels.read_montage('GSN-HydroCel-257')`.

After running `mne.io.read_raw_egi` I get this with `print(raw.info)`:

> <Info | 20 non-empty fields
>     bads : list | 0 items
>     ch_names : list | E1, E2, E3, E4, E5, E6, E7, E8, E9, ...
>     chs : list | 257 items (EEG: 255, EOG: 1, MISC: 1)
>     comps : list | 0 items
>     custom_ref_applied : bool | False
>     dev_head_t : Transform | 3 items
>     dig : list | 259 items (3 Cardinal, 256 EEG)
>     events : list | 0 items
>     file_id : dict | 4 items
>     highpass : float | 1.0 Hz
>     hpi_meas : list | 0 items
>     hpi_results : list | 0 items
>     lowpass : float | 50.0 Hz
>     meas_date : tuple | 2017-01-20 15:48:32 GMT
>     meas_id : dict | 4 items
>     nchan : int | 257
>     proc_history : list | 0 items
>     projs : list | 0 items
>     sfreq : float | 250.0 Hz
>     subject_info : dict | 0 items
>     acq_pars : NoneType
>     acq_stim : NoneType
>     ctf_head_t : NoneType
>     description : NoneType
>     dev_ctf_t : NoneType
>     experimenter : NoneType
>     gantry_angle : NoneType
>     hpi_subsystem : NoneType
>     kit_system_id : NoneType
>     line_freq : NoneType
>     proj_id : NoneType
>     proj_name : NoneType
>     xplotter_layout : NoneType
> >

I have my own MRI, generated subject BEM surfaces, but when I pull it up into `mne.gui.coregistration()`, I get something wild like this:
[image]

No matter how I try to fit the translation and rotation (Fit (ICP), Fit Fid.), I cannot match the MRI.  And when you zoom in, you can eventually see my MRI surface.
[image]

Any idea how to fix this?  I've uploaded my files (https://drive.google.com/open?id=1B58HEJQaJyT-huoyCKtkrc5ykNEolScL) if anyone wants to play around with it.

corscheid, 2019-04-10:
We do not actually have an MRI. We would have to work with a template. So far, we have been using either MNE sample or Freesurfer's fsaverage. At some point we did the mne watershed bem on this fsaverage one, if I recall correctly.

We have a plot_alignment that currently looks like this:
[image]
But we want the electrodes plotted along with this to get an idea if everything is lined up. From what I can tell by what you said, the montage file is the wrong one. Given that we have 11 Channels (FZ,F3,F4,CZ,T7,T8,PZ,P3,P4,O1,O2,HEOG,VEOG,EMG originally, but we disregard HEOG, VEOG, and EMG), which should we be using instead?

corscheid, 2019-04-12:
@agramfort 

Thank you so much for your help! It appears I got everything to plot properly by using the following lines in my script:

```python
channels = ['FZ', 'F3', 'F4', 'CZ', 'T7', 'T8', 'PZ', 'P3', 'P4', 'O1', 'O2']
montage = mne.channels.read_montage('standard_1005', ch_names=channels, transform=True)
raw.set_montage(montage)
```
[image]

I'll go back and use this on my other script as well now that everything looks okay.

agramfort, 2019-04-12:
I tried to replicate with public data with

```
import mne
from mne.datasets import sample
from mne.datasets import eegbci

data_path = sample.data_path()
subjects_dir = data_path + '/subjects'

raw_fname = eegbci.load_data(subject=1, runs=[6])[0]
raw = mne.io.read_raw_edf(raw_fname, preload=True, stim_channel='auto')
ch_names = [c.replace('.', '') for c in raw.ch_names]
raw.rename_channels({c_old: c_new for c_old, c_new in zip(raw.ch_names, ch_names)})

montage = mne.channels.read_montage('standard_1005', ch_names=raw.ch_names, transform=True)
raw.set_montage(montage)
mne.viz.plot_alignment(raw.info, subject='fsaverage', subjects_dir=subjects_dir)
```

it does not look right. What else did you do?

corscheid, 2019-04-12:
Interesting. I just copied my surfaces over and ran that and indeed got some strange alignment. Not sure what's going on there. But here is my script that yields the plots in my previous screenshot:

```python
import mne
from mne.viz import plot_alignment

# note: our project is running on macOS.
# this script is mostly a quick/dirty test to check alignments, 
# without effort to make file path literals in this code portable to Windows

subject = 'fsaverage'
subjects_dir = '/Applications/freesurfer/subjects'

# string literal filenames redacted by means of example filenames
fname_raw = 'example_raw.fif'
fname_src = 'example-src.fif'
fname_bem = subjects_dir + '/' + subject + '/bem/' + subject + '-5120-5120-5120-bem-sol.fif'
fname_trans = subjects_dir + '/' + subject + '/mri/T1-neuromag/sets/' + subject + '-trans.fif'

raw = mne.io.read_raw_fif(fname_raw)
trans = mne.read_trans(fname_trans)
src = mne.read_source_spaces(fname_src)
bem = mne.read_bem_solution(fname_bem)

raw.info['bads'] = ['EMG', 'STI 014'] 
channels = ['FZ', 'F3', 'F4', 'CZ', 'T7', 'T8', 'PZ', 'P3', 'P4', 'O1', 'O2']
montage = mne.channels.read_montage('standard_1005', ch_names=channels, path=None, unit='m', transform=True)
raw.set_montage(montage)

fig = plot_alignment(raw.info, trans=trans, subject=subject, subjects_dir=subjects_dir,
                     surfaces=['head', 'outer_skull', 'white'], coord_frame='mri',
                     meg=False, eeg=['original', 'projected'], dig=False, ecog=False, src=src, mri_fiducials=False,
                     bem=bem, seeg=False, show_axes=False, fig=None, interaction='terrain', verbose=True)
fig2 = raw.plot_sensors()
```

agramfort, 2019-04-12:
you used a trans for fsaverage. How did you produce it? can you share this
file?

>

corscheid, 2019-04-12:
I used the following bash script as a one-time setup stage. 

```bash
#!/bin/bash
set -e
set -x

export SUBJECTS_DIR="${FREESURFER_HOME}/subjects"
export SUBJECT="fsaverage"
# data filename redacted using example filename. excludes file extension.
export DATFILENAME="example"
export FWD_DESTDIR="/my/example/path"

# from http://www.natmeg.se/mne_sourceanalysis/3-Source_localization_with_MNE.html
# generate boundary element model
cd $SUBJECTS_DIR
mne_watershed_bem --overwrite

# make shortcut links
cd ${SUBJECTS_DIR}/${SUBJECT}/bem
ln -s watershed/${SUBJECT}_inner_skull_surface ${SUBJECT}-inner_skull.surf
ln -s watershed/${SUBJECT}_outer_skin_surface ${SUBJECT}-outer_skin.surf
ln -s watershed/${SUBJECT}_outer_skull_surface ${SUBJECT}-outer_skull.surf

# source space and forward model
# creates fsaverage-5120-5120-5120-bem.fif
cd -
# from https://martinos.org/mne/stable/manual/sample_dataset.html
mne_setup_mri
mne_setup_source_space --ico -6 --overwrite
mne_setup_forward_model --surf --ico 4

# copy the file where we need it
cp ${MNE_ROOT}/share/mne/mne_analyze/$SUBJECT/$SUBJECT-trans.fif \
${SUBJECTS_DIR}/${SUBJECT}/mri/T1-neuromag/sets/

# forward solution
# creates ${DATFILENAME}_eeg_raw-oct-6-fwd.fif
cd ${FWD_DESTDIR}
mne_do_forward_solution --eegonly --overwrite --mindist 5 --spacing oct-6 \
--bem ${SUBJECTS_DIR}/${SUBJECT}/bem/${SUBJECT}-5120-5120-5120-bem.fif \
--meas ${DATFILENAME}_eeg_raw.fif \
--mri ${SUBJECTS_DIR}/${SUBJECT}/mri/T1-neuromag/sets/${SUBJECT}-trans.fif
--destdir ${FWD_DESTDIR}

echo "done"
```

What is the best way to share the resulting `fsaverage-trans.fif` file?

agramfort, 2019-04-12:
E-mail Dropbox anything works

corscheid, 2019-04-12:
@agramfort I see you're using email to comment here. In case you missed it, I have edited my last response to now just say the following:

EDIT: To answer your question, it looks like I didn't actually produce this trans, but literally just copied the `fsaverage-trans.fif` provided in MNE-C `mne_analyze` folder.

agramfort, 2019-04-12:
indeed ! this works fine:

```
import os.path as op
import mne
from mne.datasets import sample
from mne.datasets import eegbci

data_path = sample.data_path()
subjects_dir = data_path + '/subjects'
trans_fname = op.join(op.dirname(mne.__file__), "data", "fsaverage", "fsaverage-trans.fif")

raw_fname = eegbci.load_data(subject=1, runs=[6])[0]
raw = mne.io.read_raw_edf(raw_fname, preload=True, stim_channel='auto')
ch_names = [c.replace('.', '') for c in raw.ch_names]
raw.rename_channels({c_old: c_new for c_old, c_new in zip(raw.ch_names, ch_names)})

montage = mne.channels.read_montage('standard_1005', ch_names=raw.ch_names, transform=True)
raw.set_montage(montage)
mne.viz.plot_alignment(raw.info, subject='fsaverage', subjects_dir=subjects_dir,
                       trans=trans_fname, eeg="projected")
```

corscheid, 2019-04-12:
Confirmed. Thank you again for your help! I suppose this Issue can now be closed.

agramfort, 2019-04-13:
see https://github.com/mne-tools/mne-python/pull/6146 to follow up.

thanks @corscheid for putting my eyes on this.

closed 2019-04-13 by agramfort