mne-tools/mne-python #6140: floating electrodes, misalignment between default montage and mri space

yvonnio, 2019-04-10:
I have my own MRI, generated subject BEM surfaces, but when I pull it up into `mne.gui.coregistration()`, I get something wild like this:
[image]

No matter how I try to fit the translation and rotation (Fit (ICP), Fit Fid.), I cannot match the MRI.  And when you zoom in, you can eventually see my MRI surface.
[image]

I've narrowed my issue down to the a potential issue with a misalignment of the default dig montage and the MRI space of my subject.  I'm using a default montage, `montage  = mne.channels.read_montage('GSN-HydroCel-257')`.

After running `mne.io.read_raw_egi` I see there are already Cardinal LPA, Nason, and RPA points set. 

`print(raw.info)`:
> <Info | 20 non-empty fields
>     bads : list | 0 items
>     ch_names : list | E1, E2, E3, E4, E5, E6, E7, E8, E9, ...
>     chs : list | 257 items (EEG: 255, EOG: 1, MISC: 1)
>     comps : list | 0 items
>     custom_ref_applied : bool | False
>     dev_head_t : Transform | 3 items
>     dig : list | 259 items (3 Cardinal, 256 EEG)
>     events : list | 0 items
>     file_id : dict | 4 items
>     highpass : float | 1.0 Hz
>     hpi_meas : list | 0 items
>     hpi_results : list | 0 items
>     lowpass : float | 50.0 Hz
>     meas_date : tuple | 2017-01-20 15:48:32 GMT
>     meas_id : dict | 4 items
>     nchan : int | 257
>     proc_history : list | 0 items
>     projs : list | 0 items
>     sfreq : float | 250.0 Hz
>     subject_info : dict | 0 items
>     acq_pars : NoneType
>     acq_stim : NoneType
>     ctf_head_t : NoneType
>     description : NoneType
>     dev_ctf_t : NoneType
>     experimenter : NoneType
>     gantry_angle : NoneType
>     hpi_subsystem : NoneType
>     kit_system_id : NoneType
>     line_freq : NoneType
>     proj_id : NoneType
>     proj_name : NoneType
>     xplotter_layout : NoneType
> >

Any idea how to fix this?  I've uploaded my files (https://drive.google.com/open?id=1B58HEJQaJyT-huoyCKtkrc5ykNEolScL) if it helps.

larsoner, 2019-04-10:
Probably one is in units of `mm` and the other `m`. Freesurfer surfaces (including head) are in `mm` units IIRC, and you probably have them in units of `m` (thus making them tiny).

yvonnio, 2019-04-10:
I think you're right.  That's a silly mistake on my part.  Would it be correct to use the scaling function in mne coreg then?

larsoner, 2019-04-10:
It would be better to create your surfaces in the proper units in the first place. But the coreg MRI scaling might also work

yvonnio, 2019-04-10:
Actually, looking at the files again with the units in mind, it seems that the info is the one to blame. 

I get the default montage by running `montage = mne.channels.read_montage('GSN-HydroCel-257')`.  Then run `raw = mne.io.read_raw_egi(
            rname, montage=montage, misc=['E257'], eog=['E31'],
            channel_naming='E%d', preload=True)` to set up the raw file.
When comparing the original montage (montage.pos) and my positions (raw.info['dig']) it seems to me the values are x1000.  Is this a bug or an issue with my scripting?

```
print(montage.pos)
[[  0.       10.56381  -2.05108]
 [ -7.82694   0.45386  -3.76056]
 [  7.82694   0.45386  -3.76056]
 [  6.96223   5.38242  -2.19061]
 [  6.48414   6.40424  -0.14004]
 [  5.69945   7.20796   1.79088]
 [  4.81093   7.77321   3.65006]
 [  3.61962   7.47782   5.50947]
 [  2.25278   6.46157   6.96317]
 [  1.18879   5.21755   8.13378]
 [  0.        3.59608   8.75111]
 [ -1.15339   1.51369   9.19904]
 [  5.94022   7.38337  -1.51513]
 [  5.07624   8.37264   0.40595]
 [  3.87946   9.03611   2.51559]
 [  2.60756   8.97868   4.39107]
 [  1.23344   8.11574   6.06161]
 [  0.        6.81181   7.28186]
 [ -1.18879   5.21755   8.13378]
 [ -2.29559   2.91372   8.5581 ]
 [  4.06489   9.40559  -0.89098]
 [  2.86784  10.01456   0.85212]
 [  1.42153  10.06322   2.84803]
 [  0.        9.40339   4.65829]
 [ -1.23344   8.11574   6.06161]
 [ -2.25278   6.46157   6.96317]
 [ -3.34467   4.40891   7.67253]
 [  1.39547  10.65281  -0.61138]
 [  0.       10.68996   1.00542]
 [ -1.42153  10.06322   2.84803]
 [ -2.60756   8.97868   4.39107]
 [ -3.61962   7.47782   5.50947]
 [ -4.49828   5.59395   6.28801]
 [  0.       10.56381  -2.05108]
 [ -1.39547  10.65281  -0.61138]
 [ -2.86784  10.01456   0.85212]
 [ -3.87946   9.03611   2.51559]
 [ -4.81093   7.77321   3.65006]
 [ -5.10466   6.41586   4.77815]
 [ -4.06489   9.40559  -0.89098]
 [ -5.07624   8.37264   0.40595]
 [ -5.69945   7.20796   1.79088]
 [ -6.16984   6.11292   3.29612]
 [ -6.01447   4.93908   4.85771]
 [ -5.33943   3.8022    6.32664]
 [ -4.64127   2.57224   7.50868]
 [ -3.53746   1.07133   8.47419]
 [ -1.99458  -0.60998   9.2887 ]
 [ -5.94022   7.38337  -1.51513]
 [ -6.48414   6.40424  -0.14004]
 [ -6.97545   5.35131   1.30741]
 [ -7.10064   4.23342   2.91874]
 [ -6.86564   3.1624    4.768  ]
 [ -6.1138    1.94213   6.23844]
 [ -5.31389   0.60081   7.48811]
 [ -3.72368  -1.14573   8.58697]
 [ -6.96223   5.38242  -2.19061]
 [ -7.31613   4.37155  -0.61128]
 [ -7.66385   3.29619   1.04415]
 [ -7.62423   2.30205   2.81799]
 [ -7.3657    1.34368   4.60382]
[... 437 lines cut ...]
>  <DigPoint |   EEG #234 : (-4124.6, -3548.0, 7954.1) mm : head frame>,
>  <DigPoint |   EEG #235 : (0.0, 3596.1, 8751.1) mm  : head frame>,
>  <DigPoint |   EEG #236 : (-2236.5, -2958.1, 8924.6) mm : head frame>,
>  <DigPoint |   EEG #237 : (0.0, -1938.3, 9458.7) mm : head frame>,
>  <DigPoint |   EEG #238 : (-7128.1, -491.9, -7349.3) mm : head frame>,
>  <DigPoint |   EEG #239 : (-7379.2, -3497.1, -2183.5) mm : head frame>,
>  <DigPoint |   EEG #240 : (-7521.8, -3700.4, -514.3) mm : head frame>,
>  <DigPoint |   EEG #241 : (-7152.1, -4711.3, 1517.6) mm : head frame>,
>  <DigPoint |   EEG #242 : (-6488.2, -5158.3, 3472.9) mm : head frame>,
>  <DigPoint |   EEG #243 : (-5530.5, -5461.8, 5501.9) mm : head frame>,
>  <DigPoint |   EEG #244 : (-4038.1, -5238.1, 7044.5) mm : head frame>,
>  <DigPoint |   EEG #245 : (-2295.1, -4878.3, 8272.2) mm : head frame>,
>  <DigPoint |   EEG #246 : (-1153.4, 1513.7, 9199.0) mm : head frame>,
>  <DigPoint |   EEG #247 : (0.0, -3741.9, 9027.9) mm : head frame>,
>  <DigPoint |   EEG #248 : (-6825.9, -1864.3, -8694.0) mm : head frame>,
>  <DigPoint |   EEG #249 : (-6740.5, -2848.4, -6747.1) mm : head frame>,
>  <DigPoint |   EEG #250 : (-6783.8, -4017.8, -5017.5) mm : head frame>,
>  <DigPoint |   EEG #251 : (-7033.5, -4450.9, -3548.9) mm : head frame>,
>  <DigPoint |   EEG #252 : (-6990.5, -5016.9, -1888.1) mm : head frame>,
>  <DigPoint |   EEG #253 : (-6675.7, -5736.1, 102.3) mm : head frame>,
>  <DigPoint |   EEG #254 : (-5968.5, -6528.6, 2032.9) mm : head frame>,
>  <DigPoint |   EEG #255 : (-5108.2, -6749.4, 3921.3) mm : head frame>,
>  <DigPoint |   EEG #256 : (-3752.2, -6677.3, 5637.2) mm : head frame>]

```

larsoner, 2019-04-10:
Probably the montage file is not in the correct units and we never correct it

massich, 2019-04-12:
yes thats exactly the case. If you look at https://11334-1301584-gh.circle-artifacts.com/0/dev/auto_examples/visualization/plot_montage.html there's the right scale for all but `EGI_256`:

```sh
fix_units = {'EGI_256':'cm', 'GSN-HydroCel-128':'cm', 'GSN-HydroCel-129':'cm',
             'GSN-HydroCel-256':'cm', 'GSN-HydroCel-257':'cm',
             'GSN-HydroCel-32':'cm', 'GSN-HydroCel-64_1.0':'cm',
             'GSN-HydroCel-65_1.0':'cm', 'biosemi128':'mm', 'biosemi16':'mm',
             'biosemi160':'mm', 'biosemi256':'mm', 'biosemi32':'mm',
             'biosemi64':'mm', 'easycap-M1':'mm', 'easycap-M10':'mm',
             'mgh60':'m', 'mgh70':'m', 'standard_1005':'m',
             'standard_1020':'m', 'standard_alphabetic':'m',
             'standard_postfixed':'m', 'standard_prefixed':'m',
             'standard_primed':'m'}
```