mne-tools/mne-python #6144: pylsl >= 1.13.1 in environment.yml

mmagnuski, 2019-04-12:
I am using mne's environment.yml file on CI and started getting errors from `pylsl>=1.13.1` requirement: 
```
Pip subprocess error:
  Could not find a version that satisfies the requirement pylsl>=1.13.1 (from versions: 1.10.3, 1.10.4, 1.10.5, 1.12.2)
No matching distribution found for pylsl>=1.13.1
```
This was introduced in #6120 but later @larsoner uncommented this requirement from [`requirements.txt`](https://github.com/mne-tools/mne-python/blob/master/requirements.txt) but it is still there in [`environment.yml`](https://github.com/mne-tools/mne-python/blob/master/environment.yml).

agramfort, 2019-04-12:
can you open a PR?

>

mmagnuski, 2019-04-12:
sure

PR #6145 (2019-04-12): fix for #6144
Fixes #6144
closed 2019-04-12 by agramfort