mne-tools/mne-python #6154: regression in vol space

virvw, 2019-04-16:
If your issue is a usage question, please consider asking on our [Gitter channel](https://gitter.im/mne-tools/mne-python) or on our [mailing list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) instead of opening an issue.

#### Describe the problem
#### Describe your solution
#### Describe an alternative

We (@virvw and @poojaprabhu9) wish to perform the mne.stats.linear_regression in source space. 
Ideally, we wish to perform these after morphing all participant data on a fsaverage stc or src so as to build to group statistics. So the idea: one linear regression (across N experimental conditions) per participant in source space then perform a group statistics on the betas. We (with @bgauthie Cereb Cortex 2018) had done something similar but only did this in sensor space and then projected the betas into source space. This time, we wish to perform this directly in source space in order to target specific brain structures (in particular hippocampus).

I can see two a priori options: 
(1) perform the regression on all vertices on the full brain (stc) and extract significant vertices p values of betas, and correct for multiple comparisons using spatiotemporal clustering).

(2) perform the regression per label (e.g. HCPMMP-1 labels segmented stc).

However, (1) and (2) do not allow to look at hippocampus. Hence:

(3) perform the regression in full volume (incl. cortex + hippocampus volumes)

Once linear regression has been conducted for all participants, we average their beta value time courses (at the level of vertices or labels) and perform a group average stat on the betas.

Currently, we don't seem to be able to do (1), (2) or (3). Have you already thought about this? If not, how can we help (eventually adding a new def in the main function)

#### Additional context
Add any other context or screenshots about the feature request here.

larsoner, 2019-04-16:
> However, (1) and (2) do not allow to look at hippocampus.

Thoughts

1. You could make a MixedSourceEstimate if you want, cortex + hippocampus. But we don't yet support different orientation priors for mixed source spaces so probably isn't worth it over just using a full volume source space.
2. Why can't you do (3)? You'll probably want to use `VectorVorlumeSourceEstimate` (`pick_ori='vector'`) in order to keep things linear. Perhaps it's too computationally intensive? 
3. You could consider using a volume source space, plus LCMV with `pick_ori='max_power'` which has the nice property that it collapses the free orientation source space into a fixed orientation one, with the orientation chosen for each voxel based on maximum power (according to the `data_cov` you provide). A drawback is that this depends on your choice of `data_cov` (in addition to the `noise_cov`). This is the approach taken in #6110 (alongside a probably much too large `spacing` arg to make the computation tractable)
4. At some point we should have a way of "label-izing" volumes. Freesurfer probably has utilities/files for this but I've never looked or worked with them (probably `aseg.mgz`), and in general we should have a way of making volume labels. Basically we need:

    1. `read_labels_from_aseg` (or whatever the correct Freesurfer file is), which returns a list of...
    2. `VolumeLabel` or some equivalent (needs suitabl API and I/O functions)
    2. `extract_label_time_course` extended to work for `VolSourecEstimate` + `VolLabel`
    3. (Eventually make everything work for mixed source spaces)

    This might be a good sprint target. I'm happy to work on this since I'm going to be working with volume source spaces more soon. I'll add an issue so people can give ideas/feedback.

jona-sassenhagen, 2019-04-16:
Algorithm wise, with our use case, regression scales well with the number of targets (sensors/voxels), the issue is usually the number of features (sampling rate/window lengths). I think this should still go with, what, 3000-6000 targets you have in source space? Worst case we can split apart the regression problem ourselves (effectively first compute (X'X)^-1, or do the most effective factorisation).

Regression per label also sounds good to me though. If labels have better signal to noise ratio and you want to regularise, this would be advantageous, right?

It should not be too hard to support both. All the functionality is reasonably well modularised.

virvw, 2019-04-16:
Sounds good, let me touch based with @poojaprabhu9 on option (3). Her first attempt was hindered when morphing volumes across participants as she seemed to obtain a variable number of vertices across participants after morphing. We have to check whether this is a coding issue or an actual bug.

jona-sassenhagen, 2019-04-27:
I totally forgot to cue in @dengemann and @JoseAlanis here.

larsoner, 2020-04-27:
@virvw I'll go ahead and close this since it seems like with volume labeling/extraction (#6155 and #7639) this should be possible. But if we run into problems with efficiency or things don't work feel free to reopen and we'll figure out how to make it work better.

closed 2020-04-27 by larsoner