mne-tools/mne-python #6155: ENH: Volume labeling

larsoner, 2019-04-16:
At some point we should have a way of "label-izing" volumes like we do for surfaces. Freesurfer probably has utilities/files for this but I've never looked or worked with them (probably aseg.mgz), and in general we should have a way of making volume labels. Basically we need (API proposal):

1. `read_labels_from_aseg` (or whatever the correct Freesurfer file is), which returns a list of...
2. `VolumeLabel` or some equivalent (needs suitabl API and I/O functions)
3. `extract_label_time_course` extended to work for `VolSourecEstimate` + `VolLabel`

Eventually we should make everything work for mixed source spaces, too, but there are probably other things to sort out before then (e.g., mixed orientation priors).

agramfort, 2019-04-16:
+100

>

vferat, 2019-08-14:
Maybe the labeling of volume sources should be done during the creation of the` SourceSpaces `instance. When using `mne.setup_volume_source_space`,  setting the `MRI` value to the specific atlas you want ( such as aprac+aseg.mgz)  should store labels for each source. 
This may be more consistent when setting the `volume_label` argument as labels from the mri file could be use to map the brain with sources.

agramfort, 2019-08-16:
given a dipole location in a vol source space you should be able to
know in why label of a parcellation it belongs to.
We should rely on nilearn as much as possible for this.

vferat, 2019-09-09:
Logically, annotations and aseg files should share the same labels names.

Shall we create a new class `VolumeLabel` or add an attribute to the existing one `Label` such as `voxels`.
 `LabeL.voxels` would be an array (n_voxels, 3) containing the x,y,z positions of each voxels belonging to the label ? 
Maybe another attribute such as  `LabeL.trans` will be needed to store the vox2ras_tkr transformation and assign sources to label later.

In `extract_label_time_course` we could check if a source belongs to a label by checking either `Label.vertices` if the source is a surface one or `Label.voxels` if the source is a `VolSourecEstimate`

agramfort, 2019-09-09:
to mimic the Label logic that is defined wrt the high resolution surface the VolumeLabel / VolLabel should be defined according to T1.mgz. I would have it use an attribute vertices (would avoid unnecessary branching I suspect) that would be a flat list if integers which corresponds to coordiantes in MRI 3d array after ravelling it to 1d. VolLabel would then be very close to Label API.

my 2c

larsoner, 2019-10-31:
See also closed #6278 where @ckiefer0 has some working scripts

larsoner, 2019-10-31:
We have for example `pca_mean`. But for new modes let's open a new issue since it's (mostly) orthogonal to deciding how to support volume labeling + extraction

larsoner, 2019-10-31:
(or pca_flip, rather)

poojaprabhu9, 2020-02-04:
We (@virvw and @poojaprabhu9) are trying to perform linear regression on the volume source estimates. We want to try regression on volume labels consisting of those structures explained this example https://mne.tools/dev/auto_examples/visualization/plot_parcellation.html. I found that the issue is open but not sure is it solved?
Any luck with this problem.
Will be of great help.

larsoner, 2020-02-04:
No progress that I know of other than the scripts linked above

poojaprabhu9, 2020-02-04:
It will be of great help, if you can resolve this.
Please.

virvw, 2020-02-04:
Dear @larsoner @agramfort - is this something we could help achieve / make progress on or is it currently out of reach? (rather decisive in how to finalize a study for @poojaprabhu9 hence her plea ;-)

larsoner, 2020-02-05:
It might take a bit of API design and work to get right. I'm not sure when I'd get to it (not in the next week or two probably).

One concrete step that might work is making `extract_label_time_course` deal with the `Vector*` variants of SourceEstimates work properly (probably just with `'mean'` mode). That way volume and other free orientation source estimates could be processed properly.

Then we could decide on a way to encode volume information properly. In the meantime here is some code that I wrote to take an atlas (in the same space as the subject) and give a binary mask for the voxels of interest:

https://gist.github.com/larsoner/a2be47b0489ac04834f67c883cc4d6c2

It contains specialized code for atlases/subjects, but hopefully it at least gives you something to use to code your own extractor. This plus the scripts from #6278 hopefully will give a start...

larsoner, 2020-04-10:
I'm wondering if the idea of creating volume labels like we currently have for surfaces is overkill/not so useful. I had a brainstorming session with @agramfort and a simpler API (to implement and use, I think) would be:

- `vol_stc.in_label(atlas, idx, src)` where for example you could do:
    ```
    aparc, lut = mne.read_freesurfer_volume_atlas('aparc.a2009s+aseg.mgz', subject, subjects_dir)
    vol_stc.in_label(aparc, lut['ctx-lh-transversetemporal'], src)
    ```
    or with your own custom atlas in the subject's space:
    ```
    vol_stc.in_label(custom_atlas, idx, src)
    ```
- `extract_label_time_course(vol_stc, atlas, src)` to obtain an ndarray of shape `(n_rois, n_times)`

Advantages:

- remains pretty readable within the Freesurfer context by allowing us to load an atlas for the given subject (along with the lookup table for region to atlas values)
- simultaneously allows `atlas` to be a generic SpatialImage (or str path to something `nib.load`-able) *that must be in the subject's MRI space* with arbitrary `idx` numbers.
- stays closer to the MRI world when working with the data, which should help integration with other packages, etc.

One potential disadvantage is that it puts the onus on the user to transform their atlas to the subject space if it isn't already there, but there are nice tools for this already outside of MNE I think, and it's not really in scope for our package (or sustainable for us) to do so.

Once simple mean over the labels is done, we can extend to other methods (like `pca`).

@virvw @poojaprabhu9 @vferat @pshahbasak @ckiefer0 WDYT? Also @britta-wstnr I think you work on volumetric data and hence might be interested in this

larsoner, 2020-04-10:
@vferat this sort of thing can also be ported over to `setup_volume_source_estimate` if you do want to make custom MixedSourceEstimates with surface cortical + volume subcortical spaces, for example (see also #7597)

vferat, 2020-04-11:
Seems alright for me:
Maybe add a generic public function to read freesurfer luts that may not be distributed with freesurfer ( here an exemple with Schaefer atlas):
```
aparc = nib.load(aparc_Schaefer2018_100Parcels_17Networks.mgz)
lut = mne.read_freesurfer_lut( Schaefer2018_100Parcels_17Networks_LUT.txt)
vol_stc.in_label(aparc, lut['17Networks_LH_LimbicA_TempPole_1'],  src)
```
Also this may requieres to  add support for `VolSourceEstimate` to `extract_label_time_course` (so far the first 2 source spaces have to be surf type ) if we want to be able to extract tc for Volume sources only. (I'm not sure it's on the scope of MNE)

Modifying  `setup_volume_source_estimate` , should be as simple as adding an input name 'lut' with default value targetting the freesurfer default LUT. A combination of custom `volume_label` and custom 'lut' will allow user to used custom atlas.

larsoner, 2020-04-11:
> Maybe add a generic public function to read freesurfer luts

Good idea, I think we have it privately already, we could make it public as part of this.

> Also this may requieres to add support for VolSourceEstimate to extract_label_time_course (so far the first 2 source spaces have to be surf type ) if we want to be able to extract tc for Volume sources only. (I'm not sure it's on the scope of MNE)

Indeed, this is (implicitly) part of #7597

> Modifying setup_volume_source_estimate , should be as simple as adding an input name 'lut' with default value targetting the freesurfer default LUT. A combination of custom volume_label and custom 'lut' will allow user to used custom atlas.

Would this be helpful for you?

vferat, 2020-04-13:
> Would this be helpful for you?

Yes, I think it's necessary if people want to use only volume sources, and reject some region such as cerebellar white matter or other unwanted region (optic chiams ect..)

I don't know how much work it is, but I could be interesting to use a `include` / `exclude` API instead of `volume_label ` for creating the volume source space. These two mutually exclusive arguments, either `list of int `or `list of string` will allow user to pick region by value in the MRI, or name in the lut.

While working with Mixed source space, `extract_label_time_course` may take both `atlas` and `labels ` arguments. If the 2 first source spaces are no longer of type Surf, and/or there are no sources in some volume `vol_stc.in_label`, time courses order could be confusing. I suggest returning the regions names (list of string) to prevent a lot of errors (by adding an input `return_names=False` for backward compatibility.

Let me know if you think it's worth, and if you need some help to work on it.

larsoner, 2020-04-13:
> I don't know how much work it is, but I could be interesting to use a include / exclude API instead of volume_label for creating the volume source space.

I'll have to think on this one, and how it interacts with `volume_label`. It seems like at least a mostly separable issue, can you open a new issue to describe the use case and your proposed API? That way we can tackle it separately and not lose track of it.

>  If the 2 first source spaces are no longer of type Surf, and/or there are no sources in some volume vol_stc.in_label, time courses order could be confusing. 

We already have an `allow_empty` kwarg for `extract_label_time_course`. We should do whatever it does for surface source space labels when there are no source vertices corresponding to a given ROI, then there is no need for a new argument or return value and we stay internally consistent.

larsoner, 2020-04-24:
FYI I have an implementation based on the discussions above in #7639. Feel free to have a look at the API and try it out!

poojaprabhu9, 2020-04-24:
I was trying to read the volume stc which were computed previously using
mne.read_source_estimate in mne.v0.21.dev0.
I met with an error
TypeError: vertices must be an instance of list, got <class
'numpy.ndarray'> instead.
without reading stcs i cannot check the extract_label_time_course ,correct?
How to solve this?

larsoner, 2020-04-25:
Should now be fixed in the PR (let's keep conversation there for now)

closed 2020-04-30 by larsoner

poojaprabhu9, 2020-05-07:
You mean to say I can use atlas in .nii format to extract the time course from stc?
(Without updating MNE)

PR #7639 (2020-04-30): MRG, ENH: Add volumetric atlas support
Adds volume label support to `extract_label_time_course`. Requires #7597 (and follow-up PRs).

- [x] Add `mne.read_freesurfer_lut` as suggested by @vferat [here](https://github.com/mne-tools/mne-python/issues/6155#issuecomment-612420561) to make `setup_volume_source_space` more general and pave the way for more generic atlas support
- [x] Speed up `setup_volume_source_space` with many labels by refactoring `_make_volume_source_space`
- [x] Add test that `setup_volume_source_space` with all volume labels of `aseg.mgz` creates a complete source space
- [x]  [... cut]