mne-tools/mne-python #6159: MAINT: Beamformer simulation study

larsoner, 2019-04-16:
To be filled in by @wmvanvliet (turning project stub into issue).

wmvanvliet, 2019-04-19:
The beamformer code is getting more powerful with every PR we push. However, this comes at the cost of accumulating parameters at a rapid pace. We need to find out which configurations work well and which do not. During the sprint, we're going to design and execute this study.

The study will be performed both on [phantom data](https://martinos.org/mne/stable/auto_tutorials/plot_brainstorm_phantom_elekta.html?highlight=phantom) and simulated data.

My idea right now is to test the LCMV and DICS beamformers with a huge grid of all possible parameter permutations and compute the performance on the phantom and simulated dataset in terms of:

 - Location of peak source activity versus true source origin
 - Field spread
 - Depth bias
 - Orientation bias

And use [multi objective optimization](https://en.wikipedia.org/wiki/Multi-objective_optimization to map out optimal configurations.

britta-wstnr, 2019-04-22:
Let's add a comparison to FieldTrip to the list, happy to work on that one.

wmvanvliet, 2019-04-22:
We're working in a separate repo for this: https://github.com/wmvanvliet/beamformer_simulation

larsoner, 2019-04-26:
Done by @wmvanvliet, the conclusions led to updating some other beamformer issues for further actions

closed 2019-04-26 by larsoner