mne-tools/mne-python #6165: remove RuntimeWarning in plot_compare_evokeds?

drammock, 2019-04-17:
```py
plot_compare_evokeds(list_of_evokeds, picks='grad')
```
yields
`RuntimeWarning: More than 6 channels, truncating title ...`

Two suggested enhancements:

1. if `picks` is one of the standard types (`eeg`, `mag`, `grad`...) then the title should reflect the channel type, rather than trying to list all channel names
2. if specific channels are requested, the truncation should happen with an info message printed to the terminal, not a `RuntimeWarning` (admittedly subjective, but I think a warning seems too strong for a "we truncated your plot title" message)

drammock, 2019-04-17:
I can take this on at the sprint; cc @jona-sassenhagen because of possible interaction with #4746

jona-sassenhagen, 2019-04-17:
Agreed!

I think it's fairly independent and I agree with your judgements though. But we will talk about all this at the sprint anyways.

commit 52949f1b (2019-04-22): MRG+1: better title in plot_compare_evokeds (closes #6165) (#6173)

* better title in plot_compare_evokeds (closes #6165)

* use logger.info instead of warn

* fix stupidity

* simplify
closed 2019-04-22 by larsoner