mne-tools/mne-python #6168: FreeSurfer and Python Interaction

jgeller112, 2019-04-17:
Hello, 

I am currently working my way through the anatomy.py script (the `def process_subject_anat` function) using the most recent version of MNE (0.18) and Spyder (Python 3.7). I ran into an error when running the script: 

 File "<ipython-input-24-cd737f5d04ec>", line 94, in <module>
    parallel(run_func(subject_id) for subject_id in range(1, 20))

  File "/Users/gellr/anaconda3/lib/python3.7/site-packages/sklearn/externals/joblib/parallel.py", line 930, in __call__
    self.retrieve()

  File "/Users/gellr/anaconda3/lib/python3.7/site-packages/sklearn/externals/joblib/parallel.py", line 833, in retrieve
    self._output.extend(job.get(timeout=self.timeout))

  File "/Users/gellr/anaconda3/lib/python3.7/site-packages/sklearn/externals/joblib/_parallel_backends.py", line 521, in wrap_future_result
    return future.result(timeout=timeout)

  File "/Users/gellr/anaconda3/lib/python3.7/concurrent/futures/_base.py", line 432, in result
    return self.__get_result()

  File "/Users/gellr/anaconda3/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception

RuntimeError: I cannot find freesurfer. The FREESURFER_HOME environment variable is not set.

The FreeSurfer environment is set in my terminal (I can run recon from it). Any ideas on how to fix this? 

Best,
Jason

jgeller112, 2019-04-17:
I also get this error: 

FileNotFoundError: [Errno 2] No such file or directory: 'recon-all': 'recon-all'

agramfort, 2019-04-18:
you need so install and setup freesurfer before running this.
recon-all is a freesurfer command line tool.

jgeller112, 2019-04-18:
FreeSurfer is installed in my computer—I am able to recon files from it. Oddly, when I open Anaconda directly from the terminal it recognizes FreeSurfer.

closed 2019-04-18 by jgeller112

agramfort, 2019-04-18:
if you use spyder you need to source freesurfer before starting it from a
terminal.

HTH

>

luoying050601, 2020-10-02:
> you need so install and setup freesurfer before running this. recon-all is a freesurfer command line tool.

sorry, can I ask a question? If I installed the freesurfer, can I run the `cortex.segment.init_subject("S" + user, anat_path)` without error_msg? Or I have to use freesurfer to do this command?(can python call the freesurfer outside the freesurfer?)