mne-tools/mne-python #6170: psd_multitaper issue in 0.18.dev0

kwdoyle, 2019-04-18:
Hello,

There is an issue with psd_multitaper which does not happen in version  0.17.0. When passing an epochs object to psd_multitaper, the following errors can occur:

```python
File "<ipython-input-340-b4b815fe6cc5>", line 33, in my_function
    psd_data, frequencies = psd_multitaper(epoch, fmin=fmin, fmax=fmax)
  File "</usr/local/lib/python3.7/site-packages/mne/externals/decorator.py:decorator-gen-72>", line 2, in psd_multitaper
  File "/usr/local/lib/python3.7/site-packages/mne/utils/_logging.py", line 89, in wrapper
    return function(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/mne/time_frequency/psd.py", line 281, in psd_multitaper
    data, sfreq = _check_psd_data(inst, tmin, tmax, picks, proj)
  File "/usr/local/lib/python3.7/site-packages/mne/time_frequency/psd.py", line 45, in _check_psd_data
    picks = _picks_to_idx(inst.info, picks, 'data', with_ref_meg=False)
  File "/usr/local/lib/python3.7/site-packages/mne/io/pick.py", line 919, in _picks_to_idx
    picks = _picks_str_to_idx(info, picks, exclude, with_ref_meg)
  File "/usr/local/lib/python3.7/site-packages/mne/io/pick.py", line 1019, in _picks_str_to_idx
    % (bad_name, bad_type))
ValueError: picks (array-like of str) could not be interpreted as channel names (no channel "data"), channel types (no type "data"), or a generic type (just "all" or "data")
```

To complicate matters, this issue does not occur on all files.

larsoner, 2019-04-18:
I can't reproduce on a first try:
```
import numpy as np
import mne
epo = mne.EpochsArray(np.zeros((1, 1, 1000)), mne.create_info(1, 1000., 'eeg'))
mne.time_frequency.psd_multitaper(epo)
```
And it's not that some of your files have channel names `data`, because this gives:
```
>>> epo = mne.EpochsArray(np.zeros((1, 1, 1000)), mne.create_info(['data'], 1000., 'eeg'))
>>> mne.time_frequency.psd_multitaper(epo)
...
  File "/home/larsoner/python/mne-python/mne/io/pick.py", line 919, in _picks_to_idx
    picks = _picks_str_to_idx(info, picks, exclude, with_ref_meg)
  File "/home/larsoner/python/mne-python/mne/io/pick.py", line 1021, in _picks_str_to_idx
    raise RuntimeError('Some channel names are ambiguously equivalent to '
RuntimeError: Some channel names are ambiguously equivalent to channel types, cannot use string-based picks for these
```
Can you make a minimal example? If you need to you can upload a hopefully small data file somewhere for us to download to reproduce.

larsoner, 2019-08-21:
No response, closing

closed 2019-08-21 by larsoner