mne-tools/mne-python #6171: picks argument to get_data

jona-sassenhagen, 2019-04-20:
Currently, if you want to do e.g. decoding, you'll do something like `X = epochs.get_data()`. Now this will often pull in the stim channel, which might e.g. lead to perfect accuracy at t=0. So you would do: `X = epochs.pick_types(stim=False).get_data()` or something like that.

This isn't too inelegant code wise, but 1. it's easy to forget, 2. it's wasteful - we construct a new Epochs object just to remove one chan! It would be more efficient if get_data could take some argument to indicate what types to pick. Then, we could simply index into the returned array, instead of constructing a completely new epochs object. E.g.:

`X = epochs.get_data(stim=False)`

API wise, we could:
- accept the same arguments as pick_types
- accept a dict of types to bools, or a collection of strings
- take a picks argument

Thoughts?

larsoner, 2019-04-20:
I would just add `picks` arg, not all the types. Types are now implicitly supported anyway

jona-sassenhagen, 2019-04-20:
Good point.

Anybody can think of a candidate for this? It should be easy.

Or I can do it at the sprint.

agramfort, 2019-04-21:
I added it to https://github.com/mne-tools/mne-python/projects/4

>

PR #6198 (2019-04-24): MRG: Picks argument to epochs.get_data 
Closes #6171

What about an exclude param (for exclude="bads")?
closed 2019-04-24 by larsoner