mne-tools/mne-python #6178: plot_compare_evokeds zooms out of axes

stfnrpplngr, 2019-04-22:
Using `mne.viz.plot_compare_evokeds` has plotting problems when using zoom. The lines are plotted outside of figure axes.

Default:
[image]

Zoom:
[image]

Here, I use some random data:
```Python
import numpy as np
import mne

rng = np.random.RandomState(0)
n_epochs, n_channels, n_times = 50, 10, 20
sfreq = 1000.
data = rng.randn(n_epochs, n_channels, n_times)
events = np.array([np.arange(n_epochs), [0] * n_epochs, np.ones([n_epochs], dtype=np.int)]).T
info = mne.create_info(n_channels, sfreq, 'eeg')
epochs = mne.EpochsArray(data, info, events)

mne.viz.plot_compare_evokeds([epochs[:int(n_epochs/2)].average(), epochs[int(n_epochs/2):].average()], picks=[0])
```

jona-sassenhagen, 2019-04-25:
I discussed with @stfnrpplngr that I think it should be very low priority to make this work as is. I would prefer if we wrote a different, more appropriate callback for click and drag here. One simple option would be plotting a copy of the cropped evokeds objects (cropped to the time interval selected by the user). Another would be to show bar plots of the mean values in the dragged window.

drammock, 2019-04-25:
I thought that the zooming was default matplotlib zooming, not something we wrote. If what we currently do isn't compatible with mpl, I'm inclined to try to fix it before adding a new callback. Let's discuss tomorrow morning.

On April 25, 2019 10:06:02 AM PDT, jona-sassenhagen <notifications@github.com> wrote:
> I discussed with @stfnrpplngr that I think it should be very low
> priority to make this work as is. I would prefer if we wrote a
> different, more appropriate callback for click and drag here. One [...]

drammock, 2019-04-26:
After several long discussions, @agramfort @jona-sassenhagen and @drammock agree that we are closing this as `[won't fix]`.  Having beautiful defaults *and* elegant cooperation with the matplotlib GUI interface is too difficult to maintain.  Sorry :(

closed 2019-04-26 by drammock