mne-tools/mne-python #6179: DOC ENH: better setup instructions for new contributors

drammock, 2019-04-22:
Currently there is some info in [the contributing page](https://mne-tools.github.io/dev/contributing.html#id1) and some other info in [the advanced setup page of the install docs](https://mne-tools.github.io/dev/advanced_setup.html#using-the-development-version-of-mne-python-latest-master) about how to set up your system as an MNE developer.  This info should only be in one place.  @jasmainak suggests we make a `CONTRIBUTING.rst` file with (more detailed) info on system setup, and use the sphinx `include::` directive to paste that file into one of the documentation pages.

@drammock has a bash script for system setup; parts of that could be added.

Questions:

1. how much content should go in `CONTRIBUTING.rst`?  (i.e., just the code to set up your dev environment?  basically all of the current contributor's guide? something in between?)

2. where in the docs should it be included?  in the contributing guidelines page, or in the advanced installation docs?

cc @massich

massich, 2019-04-22:
Apparently, we hit the issue where the documentation on how to interact with the repo itself gets duplicated between repos, outdated, etc.. 

cc: @emdupre

emdupre, 2019-04-22:
Adding a .github repository to the org might help, here !

https://github.blog/changelog/2019-02-21-organization-wide-community-health-files/

PR #6254 (2019-05-06): MRG, DOC: overhaul contributor guide
Closes #6179 

- [x] adds a new file `CONTRIBUTING.rst` to the repository root, that gives a short overview of how to contribute
- [x] overhauls `doc/contributing.rst` to give more detailed guidance and bring it up to date.
- [x] ultimately remove `doc/configure_git.rst` and `doc/customize_git.rst`; give just enough detail in `doc/contributing.rst` to get people off the ground, and point them to the many other resources for further git learning

cc @massich @jasmainak @jeythekey
closed 2019-05-06 by larsoner