mne-tools/mne-python #6185: [Feature request] Request for improvement of topography with Neuromag

Shohei-Mizutani, 2019-04-23:
#### Describe the problem

We measured MEG data with Elekta machine this year. Those data were measured with the gradiometer only, without the magnetometer.
And, when we analyzed those data of measurement results with MNE python, those topography graphs showed only positive values and drew red distributions over the entire surface. One of the possible causes is that MNE python may be outputting the RMS value of the sensor for 2 channels of X and Y axes of the flat gradiometer of Neuromag.
I would like you to solve this problem, is it possible?

We can not attach the problem data and program because the data is more than 10MB, but we can give them by e-mail etc.

agramfort, 2019-04-23:
yes it's because we display the RMS of the pairs of grads.

have a look at this
https://martinos.org/mne/dev/auto_examples/preprocessing/plot_virtual_evoked.html

if it works please close the issue.

>

larsoner, 2019-04-23:
You can also often pick one of the two grads with `meg='planar1'` or `meg='planar2'` or `picks='planar1'` or `picks='planar2'` to show just one of the two channels depending on the function(s) you're using

Shohei-Mizutani, 2019-04-23:
Thank you for your prompt answer. And thank you for the information about how to solve.

We also tried _Remap MEG channel types_, but the following value error was given at [`evoked.as_type('mag')`].

[image]

We attach the program where the problem was.
[program.zip](https://github.com/mne-tools/mne-python/files/3106303/program.zip)

I would appreciate it if you could give me about the problem.

agramfort, 2019-04-23:
you need to find a dataset with info that contains the mags information.

>

Shohei-Mizutani, 2019-04-26:
Thank you both for your guidance.

Just to clarify my understanding, it’s not possible to interpolate the grads to mags because there are no mags to start with, but if we choose just one of the 3 planar gradiometers, we should be able to display it with a red/blue color map that contains both positive and negative values.
But, this method could not display the topography.
And, however if we display just 1 of 3, the output may not be as meaningful.
Is that correct?

agramfort, 2019-05-01:
> Just to clarify my understanding, it’s not possible to interpolate the
> grads to mags because there are no mags to start with, but if we choose
> just one of the 3 planar gradiometers, we should be [...]

you can plot signed values with planar1 or planar2 gradiometers but it
won't look nice ie with dipolar patterns.

you'll need to hack your evoked object as your use case is not easily
covered.
- you need to find the measurement info for the magnetometers you miss (eg
from sample data)
- create a new evoked object with the missing magnetometers
- interpolate.

disclaimer I never tried and it may not be that easy.

maybe someone can help.

larsoner, 2023-06-09:
Closing this one as something we probably won't fix/address because it seems like a corner case, but we can reopen if it's still a blocker for people!

closed 2023-06-09 by larsoner