mne-tools/mne-python #6204: [regression on grads - info needed]

virvw, 2019-04-24:
We applying "linear_regression" in sensor space, are the grads combined prior to it? i.e. are the beta based on single grads or pairs of grads? A priori it looks like single grads so the graphic depiction of the outcomes (which reports the norm) may be misleading/confusing.

For instance: say gradx goes from negative to positive amplitude as a function of a variable of interest wereas grady goes positive to negative. They both show a significant beta. You norm, and the depiction becomes zero. (I am simplifiying just to get the gist of it)

So minimally there seems to be a bit of a discrepancy between what the stats compute and what the plotting reports. 

currently discussing this with @Nichalas

larsoner, 2019-04-24:
> are the grads combined prior to it?

They shouldn't be IIRC.

> They both show a significant beta. You norm, and the depiction becomes zero. 

The norm we use for planar gradiometer pairs is not just a simple sum (which would indeed cancel out) but rather the RMS. So if the beta for planar1 is big and positive and the beta for planar1 is big and negative the result will be big (and positive). So basically things that are far from zero remain far from zero, but you do lose the directionality.

dengemann, 2019-04-24:
Nothing is merged internally while fitting the models. It's 1 mode per sensor per time point. The plotting always combines both grads. This is ultimately what you want to be able to source localize the ensuing results.

virvw, 2019-04-24:
Thanks for the quick reply - what is  IIRC?

It partially answers but not fully. Basically, for interpretation purposes, it matters whether betas are negative or positive. When looking at the rms of the betas in grads, they will always be positive.

jasmainak, 2019-04-24:
> what is IIRC?

If I remember correctly (IIRC)

welcome to github ;-)

larsoner, 2019-04-24:
FYI (for your information :) ) you can also use `meg='planar1'` or `meg='planar2'` in some/most picking and viz functions to look at just one grad type, which avoids the norm. I'll close for now but let us know if these ideas won't work for your use cases

closed 2019-04-24 by larsoner