mne-tools/mne-python #6219: extra "misc" row in raw.plot(butterfly=True)

drammock, 2019-04-25:
hey @stfnrpplngr, since you're working on butterfly mode, can you confirm this?

```py
import os
import mne
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, preload=False, verbose=False)
raw.crop(10).load_data()
raw.pick_types(meg=False, eeg=True).plot(butterfly=True)
```
I'm seeing an extra row for "misc" channels even though I've done `pick_types`

cc #6210

larsoner, 2019-04-25:
FWIW the epochs butterfly mode and raw butterfly mode use slightly different code paths, so this might not be trivial to fix in #6210 :(

drammock, 2019-04-25:
OK, I'll self-assign then. @stfnrpplngr if you want to take it on before I get around to it, LMK.

stfnrpplngr, 2019-04-25:
Applying my changes in epochs.py to raw.py will probably solve the problem. I will do that after finishing the refactoring of epochs.py.

PR #6229 (2019-04-25): MRG: avoid empty misc row in raw butterfly plots
closes #6219 

## BEFORE

```py
import os
import mne
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False)
raw.crop(10).load_data().pick_types(meg=False, eeg=True).plot(butterfly=True)
```

[image]

## AFTER

(same code as above)

[image]

## MAKE SURE WE DIDN'T BREAK SOMETHING

```py
import os
import mne
sample_data_folder = mne.datasets.sample.data_pa [... cut]
closed 2019-04-25 by larsoner