mne-tools/mne-python #6223: Fix the cited_mne.py script to update "Publications by users" page

sappelhoff, 2019-04-25:
In our docs, we link to a [Publications by users](http://martinos.org/mne/stable/cited.html) page, which is being created by our [`cited_mne.py`](https://github.com/mne-tools/mne-python/blob/master/doc/sphinxext/cited_mne.py) script, authored by @jasmainak (see https://github.com/mne-tools/mne-python/issues/1935 and https://github.com/mne-tools/mne-python/pull/2954 )

This script has not been run in a long time and according to @larsoner, it's not trivial to get it to work again (python 2 and several dead dependencies).

Some volunteer should overhaul `cited_mne.py` so that we can update our **Publications by users** and show the world how great MNE-Python is.

(the alternative is a stale page in the docs, last updated 2017)

I have not yet found a trivial workflow used by others ... or a package that can easily be tweaked / copied over.

cc: @jona-sassenhagen who tried this before

agramfort, 2019-04-25:
I would not spend time on this and point to the scholar page :)

sappelhoff, 2019-04-25:
I agree, no need to list the Gscholar output again on our website --> it's not very readable anyways.

see my PR to #6224

PR #6224 (2019-04-25): remove outdated gscholar parser and directly point to gscholar instead
closes #6223
closed 2019-04-25 by larsoner