mne-tools/mne-python #6226: Autoscale is not working properly in epochs.plot()

stfnrpplngr, 2019-04-25:
#### Describe the bug
Autoscale does not appear to work properly. It seems that scales are not calculated channel-type-wise. Probably related to https://github.com/mne-tools/mne-python/issues/5760.

#### Code to reproduce
```Python
import os.path as op
import mne

data_path = op.join(mne.datasets.sample.data_path(), 'MEG', 'sample')
raw = mne.io.read_raw_fif(
    op.join(data_path, 'sample_audvis_raw.fif'), preload=True)
raw.pick_types(eeg=True, meg=True)
raw.load_data().filter(None, 9, fir_design='firwin')
raw.set_eeg_reference('average', projection=True)  # set EEG average reference
event_id = {'auditory/left': 1, 'auditory/right': 2, 'visual/left': 3,
            'visual/right': 4, 'smiley': 5, 'button': 32}
events = mne.read_events(op.join(data_path, 'sample_audvis_raw-eve.fif'))
epochs = mne.Epochs(raw, events, event_id=event_id, tmin=-0.2, tmax=.5)

epochs.plot(scalings='auto')
```

#### Expected results
`epochs.plot(scaling=None)`
[image]

#### Actual results
`epochs.plot(scaling='auto')`
[image]

#### Additional information
Platform:      Windows-10-10.0.16299-SP0
Python:        3.7.3 (default, Mar 27 2019, 17:13:21) [MSC v.1915 64 bit (AMD64)]
Executable:    C:\Continuum\anaconda3\pythonw.exe
CPU:           Intel64 Family 6 Model 142 Stepping 10, GenuineIntel: 8 cores
Memory:        15.9 GB

mne:           0.18.dev0
numpy:         1.16.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.2.1
matplotlib:    3.0.3 {backend=Qt5Agg}

sklearn:       0.20.3
nibabel:       Not found
mayavi:        Not found
cupy:          Not found
pandas:        0.24.2
dipy:          Not found

agramfort, 2019-04-25:
the "auto" explanation is not clear here
https://martinos.org/mne/stable/generated/mne.Epochs.html#mne.Epochs.plot
I agree that it should be done per channel type.

>

stfnrpplngr, 2019-04-25:
Just checked it. `mne.viz.utils._compute_scalings()` actually is doing the scaling per channel type. The bug seems to be somewhere else.

stfnrpplngr, 2019-04-30:
PR #6255 should solve the issue. 

[image]

PR #6255 (2019-04-30): Fix _compute_scalings() for autoscale
During reshaping of the axes channel data got dispersed.

#### Reference issue
Fixes #6226.

#### What does this implement/fix?
Axes need to be swapped to ensure correct reshaping according to channel attributions.
closed 2019-04-30 by agramfort