mne-tools/mne-python #6235: Cannot pickle mne.preprocessing.ica.ICA anymore

cbrnr, 2019-04-26:
With the current master (installed via `pip install git+https://github.com/mne-tools/mne-python.git#egg=mne`), I get an error when pickling an `mne.preprocessing.ica.ICA` object:

```
import pickle
import mne

print(mne.__version__)
ica = mne.preprocessing.ICA()
raw = mne.io.read_raw_edf("/Users/clemens/Downloads/testfiles/S001R04.edf", preload=True)
ica.fit(raw)
pickle.loads(pickle.dumps(ica))
# Traceback (most recent call last):
#   File "/Users/clemens/Repositories/mnelab/.venv/lib/python3.7/site-packages/IPython/core/interactiveshell.py", line 3296, in run_code
#     exec(code_obj, self.user_global_ns, self.user_ns)
#   File "<ipython-input-8-a056623ee708>", line 1, in <module>
#     pickle.loads(pickle.dumps(ica))
# TypeError: __new__() missing 1 required positional argument: 'val'
```

The same code works with the latest released version 0.17.2. I didn't see any change to the `ICA` object that might be causing this (but maybe I'm missing something). If indeed nothing changed, could this behavior be due to how pip installs stuff from GitHub (vs. "proper" packages from PyPI)? If so, how can I fix this (I need to pickle because I want to use the `multiprocessing` module)?

agramfort, 2019-04-26:
that's weird...

jona-sassenhagen, 2019-04-27:
Is this 100% reproducible on a diff, fresh computer?

Uh ... why can't you write the ICA objects to disk the normal way?

Teekuningas, 2019-04-27:
I would blame the _Bunch_ class..

agramfort, 2019-04-28:
here is a full snippet to replicate the pb:

import mne
import pickle
ica = mne.preprocessing.ICA()
raw = mne.io.read_raw_edf(mne.get_config('MNE_DATA') +
"/MNE-eegbci-data/physiobank/database/eegmmidb/S001/S001R04.edf",
preload=True)
ica.fit(raw)
pickle.loads(pickle.dumps(ica))

any help welcome.

Teekuningas, 2019-04-28:
https://docs.python.org/3/library/pickle.html#object.__getnewargs__

You could add something like this to utils/_bunch.py 's NamedInt:
` 
    def __getnewargs__(obj):
        return obj._name, int(obj)
`

cbrnr, 2019-04-29:
Thanks @Teekuningas, this solves the problem!

PR #6252 (2019-04-29): [MRG] Fix pickling of NamedInt and NamedFloat
Fixes #6235.
closed 2019-04-29 by larsoner