mne-tools/mne-python #6250: speed up iter_evoked?

jona-sassenhagen, 2019-04-28:
I (mis)use epochs.iter_evokeds for confidence intervals from plot_compare_evokeds a lot. That is really slow. Is there some trivial way for speeding up iter_evokeds? (Instead of doing the sensible thing and speeding up plot_compare_evokeds itself ...)

I don't see one, it seems the problem is it's calling Epochs.__next__ and creating 100s of Evokeds via EvokedArray, which is simply inherently slow.

mmagnuski, 2019-04-28:
I don't know how `iter_evokeds` works exactly but I assume it could create the Evoked object only once and only change the `.data` then (if it doesn't do this already).

agramfort, 2019-04-29:
it's very possible it does a copy of the info which can take time. one
option would be to add a parameter to iter_evokeds method to avoid as many
copies as possible

jona-sassenhagen, 2019-05-04:
Thanks, might be worth doing. I will self assign, although this is low priority I think because I'm abusing things here.

agramfort, 2019-12-28:
see https://github.com/mne-tools/mne-python/pull/7164

PR #7164 (2020-01-06): MRG: Add copy param to iter_evoked
closes #6250

cc @jona-sassenhagen
commit c4d352ad (2020-01-06): MRG: Add copy param to iter_evoked (#7164)

* add copy param to iter_evoked. fix #6250

* fix

* FIX: Flake and data copy

* update latest.inc

* change default
closed 2020-01-06 by agramfort