mne-tools/mne-python #6258: `mne.events_from_annotations` returns events that barely overlap with annotation

hubertjb, 2019-05-01:
I noticed that `mne.events_from_annotations` sometimes returns events that barely even overlap with an actual annotation when `chunk_duration` is used. I think this might have been the actual reason behind the issue https://github.com/mne-tools/mne-python/issues/5684#issuecomment-438211200.

The issue seems to arise when an annotation has a duration that is only slightly over `chunk_duration`. A first event will cover almost the entire annotation, but a second annotation will be created that might start within the annotation, but will barely overlap with it. Because of that, in some rare cases, the second event is identical to the next annotation - causing `mne.Epochs` to throw an error due to duplicate events.

#### Steps and/or code to reproduce

```Python
import mne
import numpy as np

# Create fake data with annotations
data = np.random.randn(2, 16000)
info = mne.create_info(
    ch_names=['EEG1', 'EEG2'], ch_types=['eeg', 'eeg'], sfreq=256)
raw = mne.io.RawArray(data, info)

onset = [0, 30]
duration = [30.1, 30]
description = ['0', '0']
annots = mne.Annotations(onset, duration, description)
raw = raw.set_annotations(annots)

# Extract epochs
events, event_ids = mne.events_from_annotations(raw, chunk_duration=30)
epochs = mne.Epochs(
    raw, events, tmin=0, tmax=30, baseline=None, on_missing='warning')
```

#### Expected results
I would have expected `events` to contain two events only: one for the first annotation [0, 30.1] and one for the second [30, 60]. An option would be to have an argument that lets the user decide whether events need to completely overlap with the annotation to be returned.

#### Actual results

`events` is:
```
array([[   0,    0,    1],
       [7680,    0,    1],
       [7680,    0,    1]])
```

Because of the duplicate row, `mne.Epochs` returns `RuntimeError: Event time samples were not unique`.

#### Additional information
```
Platform:      Linux-4.17.13-041713-generic-x86_64-with-debian-buster-sid
Python:        3.6.6 |Anaconda, Inc.| (default, Jun 28 2018, 17:14:51)  [GCC 7.2.0]
Executable:    /home/hubert/miniconda3/envs/dl-eeg/bin/python
CPU:           x86_64: 4 cores
Memory:        15.6 GB

mne:           0.18.dev0
numpy:         1.16.0 {blas=openblas, lapack=openblas}
scipy:         1.1.0
matplotlib:    2.2.3 {backend=TkAgg}

sklearn:       0.20.2
nibabel:       Not found
mayavi:        4.6.2 {qt_api=pyqt5, PyQt5=5.11.3}
cupy:          Not found
pandas:        0.23.4
dipy:          Not found
```

agramfort, 2019-05-02:
@massich can you have a look? thx

mmagnuski, 2019-05-02:
I see how that is counter-intuitive and even wrong when one uses the events for epoching using `chunk_duration` as epoch length. The general question is: what degree of error for the chunks do we allow by default. If none then some may also find it surprising that having an annotations of duration 29.9 and requesting chunk duration of 30. they get no events.
The simplest scenario is to add something like `min_chunk_duration` and after this:
https://github.com/mne-tools/mne-python/blob/master/mne/annotations.py#L870
check the last event distance from annotation onset + duration. This would have to be `<= min_chunk_duration` to be retained.

mmagnuski, 2019-05-02:
A temp fix is to add something like:
```python
good_events = annot['onset'] + annot['duration'] - _onsets >= annot['duration']
if good_events.any():
    _onsets = _onsets[good_events]
```
before https://github.com/mne-tools/mne-python/blob/master/mne/annotations.py#L873

hubertjb, 2019-05-02:
Thanks for looking at this @mmagnuski!

closed 2019-05-04 by agramfort

hubertjb, 2019-05-04:
@mmagnuski @agramfort Thanks again!

PR #6260 (2019-05-04): MRG: fix annot->events chunk size
Closes #6258

Could also allow additional argument - `min_chunk_duration` (a bit shorter than `chunk_duration` by default for example 85%).
But in its current shape it seems to be simple enough. I could add warnings if some annotations cannot generate events due to chunk_size being too long.

- [x] add test
- [x] add warning when some annotations do not contribute any events?
- [x] decide whether to expose control over min chunk duration