mne-tools/mne-python #6259: meas_date incorrect for edf

bdyetton, 2019-05-01:
Hi MNE team, 
I believe I found a small bug. 

#### Describe the bug
When reading raw edf, info['meas_date'] is incorrect.

#### Steps and/or code to reproduce
Using another edf reader ([polyman](https://sites.google.com/view/diegoalvarezestevez/projects/polyman), [wonambi](https://wonambi-python.github.io/)) the "startdate" for the linked recording is 08-JUL-2014 23:05:09, however, when loading into mne:
```python
import mne
import datetime
edf = mne.io.read_raw_edf('example1_sleep_rec_base.edf')
print(datatime.datetime.fromtimestamp(edf.info['meas_date'][0])) #2014-07-08 16:05:09
```

[example1_sleep_rec_base.zip](https://github.com/mne-tools/mne-python/files/3136107/example1_sleep_rec_base.zip)

#### Expected results
Expected meas_date to be 2014-07-08, 23:05:09

#### Actual results
The incorrect start time of 2014-07-08 16:05:09 is given (i.e. 7 hours eariler)

#### Additional information
The offset is variable for different files, and I've yet to track down where it comes from (does not seem to be the length of recording)

larsoner, 2019-05-01:
Could one be UTC and the other local time?

bdyetton, 2019-05-01:
I don't think so. 
I am sure that 08-JUL-2014 23:05:09 is correct (these are sleep recordings). 
Are you saying that MNE converts to local time or to UTC?

larsoner, 2019-05-01:
Meas date is probably UTC

bdyetton, 2019-05-02:
Ahh, yes, nice one!
The correct time saved in the edf header can be recovered with:
```python
datatime.datetime.fromtimestamp(edf.info['meas_date'][0])+ timedelta(seconds=time.altzone)
```

closed 2019-05-02 by bdyetton