mne-tools/mne-python #6267: BrainVision annotation parsing is needlessly complicated

jona-sassenhagen, 2019-05-03:
It used to be BV files would neatly map stem codes to integers: `'Stimulus/S 11'` would become `11` and so on. Now, we have essentially random assignment.

Now it is trivial to write a small function to get neater mappings, but 1. if you work with BV files, you will probably write that same function *every single time you import your data*, which is another 5 or so lines of essentially boilerplate, and 2. this presents a significant barrier of entry to newcomers.

Of course, it could be quite easy: least-surprise default mapping BV markers to integer codes in the way you'd expect, i.e., `'S 11'` becomes `11`.

Or am I missing something?

@cbrnr @massich

cbrnr, 2019-05-04:
I haven't tested this recently, but if it is not mapped as you say it should a little function that maps events correctly would be great!

jona-sassenhagen, 2019-05-04:
I guess I should have been more involved in the original decision. And maybe I just misunderstand the intended workflow. But the current state of the art seems very suboptimal to me.

Is it documented somewhere how you should read in events from BV data?

I think we really want the whole thing to work so you can straight-forwardly get from raw to epochs. Something like

```
    event_id = {"A/X": 1, "B/X": 2, "A/Y": 11, "B/Y": 12} 
    events, _ = mne.events_from_annotations(raw)
    epochs = mne.Epochs(raw, events, event_id)
```

Simply on virtue of "Stimulus/S 1" ending up as event code 1, etc.

jona-sassenhagen, 2019-05-04:
I will do a PR as soon as possible. But I would like to hear first if I am missing some functionality that makes this a non-issue (in which case we should still doc it!).

agramfort, 2019-05-04:
you can pass event_id to events_from_annotations to choose how the
mapping is done.

does it address your problem?

jona-sassenhagen, 2019-05-04:
The issue is: the event_id I can pass to `events_from_annotations` is of str -> int form right? But how do I know what ints I should map to? Look at this real example:

```
mne.events_from_annotations(raw)[1]

{'New Segment/': 1,
 'Stimulus/S 41': 2,
 'Stimulus/S 88': 3,
 'Stimulus/S 52': 4,
 'Stimulus/S 34': 5,
 'Stimulus/S 51': 6,
 'Stimulus/S 33': 7,
 'Stimulus/S 42': 8,
 'Stimulus/S 11': 9,
 'Stimulus/S 22': 10,
 'Stimulus/S 21': 11,
 'Stimulus/S 12': 12,
 'Stimulus/S255': 13,
 'Stimulus/S 43': 14,
 'Stimulus/S 44': 15}
```

Let's say S 44 (I.e., my presentation program sending trigger code 44) is condition A/X. So now I have to write `event_id = {"A/X": 15}`, right? And I have to manually check every single of these mappings by running `mne.events_from_annotations` without event_id and checking the outcome first, *for every single subject*, right?

More generally, why would we ever map S 44 to 15, and not 44?

agramfort, 2019-05-04:
ok then event_id as a callable should be used. I am open to suggestions to
make it easier.
There was in previous release a function to do this:

https://github.com/mne-tools/mne-python/blob/maint/0.17/mne/io/brainvision/brainvision.py#L359

jona-sassenhagen, 2019-05-04:
Yes, and that function, or something like it, should have been the default for mapping markers to event code ints.

I will go about that, unless @massich or somebody else wants to.

jona-sassenhagen, 2019-05-04:
One option would be allowing `mne.events_from_annotations(raw, event_id='brainvision')`, which would automatically trigger https://github.com/mne-tools/mne-python/blob/maint/0.17/mne/io/brainvision/brainvision.py#L359 . But that seems far from optimal.

I think by default, if you do

```
raw = mne.io.read_raw_brainvision(fname)
events, event_id = mne.events_from_annotations(raw)
```

you should end up with an event_id like `{"Stimulus/S 11": 11}`, without *any* other step. Anything else seems unacceptable.

I think this requires adding something to the annotations during reading in the BV file, like `event_code`, which is per default accessed by `events_from_annotations` if present.

jona-sassenhagen, 2019-05-05:
Reading the code, I think I have a strong preference for adding an optional 5th entry to annotations: event_code (int). Every raw reader could then populate this appropriately: i.e., Brainvision triggers are very stereotypical, so they could use the sensible least-surprise parser. I'm sure there are other file formats which also have stereotypical event markers.

Another option is to add metadata to annotations objects specifying the source. `events_from_annotations` could check this information and default to a sensible parser if it finds the annotations come from Brainvision.

Distant third is `events_from_annotations(raw, event_id='brainvision')`.

@massich @agramfort @mmagnuski @cbrnr

agramfort, 2019-05-05:
I don't want to make the Annotations class smarter than it is.

I would be ok with:

1/

events_from_annotations(raw, event_id='brainvision').

or maybe

2/

events_from_annotations(raw, event_id='auto')

and if raw isinstance RawBrainVision we could default to the automagic
brainvision option.

but I feel that this function
https://github.com/mne-tools/mne-python/blob/maint/0.17/mne/io/brainvision/brainvision.py#L359
was overly complicated.

cbrnr, 2019-05-06:
I am also -1 on extending `Annotations`. I'd prefer @agramfort's option 2 with defaulting to `event_id='auto'`so that eventually the user could just do `events_from_annotations(raw)` and everything just works as expected.

jona-sassenhagen, 2019-05-06:
Yes I forgot raw instances can be traced back to their source by their subclasses. I agree 2 is the best option.

jona-sassenhagen, 2019-05-06:
I will implement it as such!

massich, 2019-05-15:
ohhh I got late to the party!! 

The idea here was to use `callable` and avoid black magic. Either by forcing the user to write her own function or passing a reader public function. This allows us to have all this logic in a function that we can document in an isolated manner, we can test the heck out of it, and educate our users (its an important thing. It has its own public function). And if there is more than one 'standard' remains clean (or revisions of a single standard).

Imagine this unreal scenario where we have two branvision vendors:

```py

from mne.io.brainvision import vendor_foo_event_parser, vendor_bar_event_parser

events, present_event_id = events_from_annotations(raw, event_id=vendor_foo_event_parser)
```
and somewhere else in the code we could be using `bar`'s specification.

having an `'auto'` conflicts with the fact that if it's None` already has a default behavior. This can be overcome by some logic. But then this logic should depend on `raw` (or the class by this matter). But then if I want the overall default behaviour which is: just give me numbers; I'll have to use a caller.

massich, 2019-05-15:
@jona-sassenhagen based in your first example what is preventing you to do something like this:

```py
    event_id = {"A/X": 1, "B/X": 2, "A/Y": 11, "B/Y": 12} 

    # work with a
    events_a, drop_a = mne.events_from_annotations(raw_a, event_id=event_id)
    epochs_a = mne.Epochs(raw_a, events_a, event_id)

    # work with b
    events_b, drop_b = mne.events_from_bnnotations(raw_b, event_id=event_id)
    epochs_b = mne.Epochs(raw_b, events_b, event_id)

    assert drop_a == drop_b  # This is not granted!
```

To my understanding, `event_id` would be experiment dependent and I might want to tune it to my needs.

jona-sassenhagen, 2019-05-15:
@massich during a workshop, I did some live coding on files the attendants had brought and ended up with this:

```
{'New Segment/': 1,
 'Stimulus/S 41': 2,
 'Stimulus/S 88': 3,
 'Stimulus/S 52': 4,
 'Stimulus/S 34': 5,
 'Stimulus/S 51': 6,
 'Stimulus/S 33': 7,
 'Stimulus/S 42': 8,
 'Stimulus/S 11': 9,
 'Stimulus/S 22': 10,
 'Stimulus/S 21': 11,
 'Stimulus/S 12': 12,
 'Stimulus/S255': 13,
 'Stimulus/S 43': 14,
 'Stimulus/S 44': 15}
```

And that is *completely* unacceptable. This is just incredibly off-putting to newcomers. This should absolutely work properly out of the box without any user input.

How we do that behind the scenes is to me not all that important at this stage, but it must work.

larsoner, 2019-05-15:
Since BrainVision is a bit special, how about we:

1. Extract numbers from `int(my_str[-3:].strip())`
2. Check that these are unique
3. If so, use them
4. If not, and/or for all that cannot be converted, use consecutive non-assigned integers to the `sorted(my_strings_without_unique_convertable_numbers)`

jona-sassenhagen, 2019-05-15:
>Check that these are unique

I think we have to account for Response events. I don't use them myself, but they are a standard feature. I.e., we shouldn't return essentially random numbers if we have both `R 11` and `S 11`.

larsoner, 2019-05-15:
Argh forgot about that. We used to have the response trig shift param, but that's presumably gone now. If we know the stim codes have some limit (probably < 1000?) maybe we should just always shift responses by 1000.

jona-sassenhagen, 2019-05-15:
I think it's always 1-255 (8 bit triggers).

jona-sassenhagen, 2019-05-15:
>maybe we should just always shift responses by 1000.

That's what may PR is doing right now ;)

massich, 2019-05-15:
If they are regular, we have no problem. Just define the mapping in a dictionary. (or something that generates something similar to this dictionary based in a regex)

jona-sassenhagen, 2019-05-15:
>If they are regular, we have no problem. Just define the mapping in a dictionary. (or something that generates something similar to this dictionary based in a regex)

You mean like

```
event_id = {'Stimulus/S{:>3}'.format(code): code for code in range(1, 256)}
event_id.update({'Response/R{:>3}'.format(code): 1000 + code for code in range(1, 256)})
```

massich, 2019-05-15:
I'm not sure if this code works but yes. And if we have extra integers we can do a counter for whatever id is not accounted for. But we need to make sure that we have the same behavior we had in 0.16. and if needed backport this to 0.17.x

then this could be exposed like this if wanted:

```py
from mne.io.brainvision import event_id

events, events_ids_present_in_raw = events_from_annotations(raw, event_id=event_id)
```

massich, 2019-05-15:
I could not see where were we testing that in 0.16. I can make you a proposal tomorrow morning.

jona-sassenhagen, 2019-05-15:
>then this could be exposed like this if wanted:

No. It must by default be reasonable.

>I could not see where were we testing that in 0.16.

We had tests like:

https://github.com/mne-tools/mne-python/blob/maint/0.15/mne/io/brainvision/tests/test_brainvision.py#L373

larsoner, 2019-05-15:
@jona-sassenhagen do you want to try restoring that test, but with modern calls / the annotations way of doing things, to see how much modification it takes? I guess the goal should be that your PR gives what you would have gotten with `response_trig_shift=1000` before, right?

jona-sassenhagen, 2019-05-15:
>I guess the goal should be that your PR gives what you would have gotten with response_trig_shift=1000 before, right?

I think that would be sufficient, yes.

>@jona-sassenhagen do you want to try restoring that test, but with modern calls / the annotations way of doing things, to see how much modification it takes?

It should pass as is with my PR, I think we're at the "make the code not bad" stage, not the "make the code do what it's supposed to do" stage (with the unresolved issue of missed triggers).

larsoner, 2019-05-15:
> I think we're at the "make the code not bad" stage, not the "make the code do what it's supposed to do" stage 

From the looks of it this old test was more comprehensive than the ~3 lines that are in your PR. I'll see how hard it is to adapt it and push a commit if it's not bad.

jona-sassenhagen, 2019-05-15:
No I mean, my code should (mostly) pass that test. I can do that too, although I must admit I have a few other things on the schedule for today still ...

larsoner, 2019-05-15:
> No I mean, my code should (mostly) pass that test

Excellent. I'll adapt it and push a commit then

massich, 2019-05-16:
>We had tests like:
/0.15/mne/io/brainvision/tests/test_brainvision.py@maint#L373

you are completely right. we checked only that the de descriptions were parsed properly. Never the translation.

jona-sassenhagen, 2019-05-16:
>you are completely right. we checked only that the de descriptions were parsed properly. Never the translation.

To be fair, I really should have been more involved with the annotations PR, to avoid this kind of issue. I was lazy there.

massich, 2019-05-16:
that's completely fine. we improved the code, and while doing so we enter a regression. We fix it, backport it done. That's the name of the game.

The only thing we need to do is to agree on what happens with the things that we have in the test suit. And if this is incomplete, complete it.

The main problem is that I simplified the test and only make sure that I recover the proper description here:
https://github.com/mne-tools/mne-python/blob/b31e573bd1bb97ab8daf6e813a6f5fa77ba266c4/mne/io/brainvision/tests/test_brainvision.py#L443-L448

now, we need to agree on what should be the translation of all these descriptions and how do we make it be as much close to 0.15 as possible. I don't think we have any issue in mapping anything we want (by the end of the day we are using `int64` so the range is -9223372036854775808 to 9223372036854775807)

```py
In [8]: events = mne.find_events(raw)                                                                                                      

In [9]: events.dtype                                                                                                                       
Out[9]: dtype('int64')
```

Then we can discuss the auto / no auto. But our parsing function should work at least for the annotations we already have in testsuit without dropping them.

massich, 2019-05-16:
I guess that what I'm not ok is with 'auto' doing something like this:

```py
description = ['New Segment/',
                'Stimulus/S253', 'Stimulus/S255', 'Stimulus/S254',
                'Stimulus/S255', 'Stimulus/S254', 'Stimulus/S255',
                'Stimulus/S253', 'Stimulus/S255', 'Response/R255',
                'Stimulus/S254', 'Stimulus/S255',
                'SyncStatus/Sync On', 'Optic/O  1']

def _bv_parser(description, add_for_r=1000):
    """Parse BrainVision event codes (like `Stimulus/S 11`) to ints."""
    try:
        event_type, code = description[:-3], int(description[-3:])
        if event_type in ("Response/R", "Stimulus/S"):
            if event_type == "Response/R":
                code += add_for_r
            return code
    except ValueError:  # probably not coercable to int
        warn("Marker '%s' could not be unambiguously mapped to an int "
             "and will be dropped." % description)

[_bv_parser(_) for _ in description]
```
It warns for things that it might be important (anything that does not end with a number) and drops silently events that do finish with a number.

```sh
<ipython-input-231-655b4a5a8885>:11: RuntimeWarning: Marker 'New Segment/' could not be unambiguously mapped to an int and will be dropped.
  "and will be dropped." % description)
<ipython-input-231-655b4a5a8885>:11: RuntimeWarning: Marker 'SyncStatus/Sync On' could not be unambiguously mapped to an int and will be dropped.
  "and will be dropped." % description)

[None, 253, 255, 254, 255, 254, 255, 253, 255, 1255, 254, 255, None, None]
```

jona-sassenhagen, 2019-05-16:
Ah right, that's an oversight.

massich, 2019-05-16:
So my execution plan would have been:
- modify the following as INTEGRATION test:
https://github.com/mne-tools/mne-python/blob/b31e573bd1bb97ab8daf6e813a6f5fa77ba266c4/mne/io/brainvision/tests/test_brainvision.py#L436 
- create a general public function in `mne.io.brainvision` that allows mapping of the signals + create UNIT test for it 
- link this function with proper defaults to brainvision raws (in a manner that it can be overwritten)

massich, 2019-05-16:
@jona-sassenhagen is there a better description about brainvision markers than this one https://pressrelease.brainproducts.com/markers/ ??

PR #6326 (2019-05-17): MRG: Brainvision events_from_annotations
Closes #6267
Closes #6276
closed 2019-05-17 by massich