mne-tools/mne-python #6268: ENH: change psd method used in ica.plot_properties from multitapering to welch

mmagnuski, 2019-05-03:
I currently have to wait some time after clicking a component in `ica.plot_components` window and the time seems to be mostly from waiting for the multitaper psd estimation to complete. I think that we should use something much simpler like welch or even fft as the default. The plotted spectrum is for informational reasons, not for analysis. I can compare timings on one of my datasets.

larsoner, 2019-05-03:
Sounds good to me

cbrnr, 2019-05-04:
OK, I would be interested in the timings if this is really the bottleneck.

jona-sassenhagen, 2019-05-04:
Sure, sounds good.

jona-sassenhagen, 2019-05-04:
While you're there, generalise plot_properties to channels please :D

agramfort, 2019-05-04:
+1 for doing a comparison

>

mmagnuski, 2019-05-04:
I profiled plot_properties and it seems I was wrong: a lot of time (> 2 s in this case) is spent in bootstrap CI computation:
[image]
We should consider switching to analytic (and possibly one-sided to better represent skewed distributions).
BTW - 7 seconds here is an exaggeration, 2 seconds were spent in plt_show before I closed the figure.

mmagnuski, 2019-05-04:
but I still think multitaper should be changed to somthing lighter (welch or fft), people with less experience with multitapersing could be mistaken by the fatter, more square-like peaks that can sometimes arise due to frequency smoothing.

mmagnuski, 2019-05-04:
@jona-sassenhagen I can take a stab at that if you insist. :)

jona-sassenhagen, 2019-05-04:
>I profiled plot_properties and it seems I was wrong: a lot of time (> 2 s in this case) is spent in bootstrap CI computation:

Uh oh, it's my fault :)

For the tests, we can pass a different ci function to plot_compare_evokeds here. You wanna do that, or should I?

>@jona-sassenhagen I can take a stab at that (generalizing to channels) if you insist. :)

I insist.

mmagnuski, 2019-05-04:
> Uh oh, it's my fault :)

This is what you meant in #6250, right?

> For the tests, we can pass a different ci function to plot_compare_evokeds here. You wanna do that, or should I?

I can test later today.

> We can also default to a different function for the default for plot_properties.

That could be useful.

> I insist.

😅

jona-sassenhagen, 2019-05-04:
>This is what you meant in #6250, right?

It is connected to that, yes. Another reason for speeding up epochs iteration.

Maybe it would also make sense to see if we can speed up the bootstrap ci function.

jona-sassenhagen, 2019-05-04:
@mmagnuski I suggest to, in `plot_ica_properties`, add a `ts_args` param and default the time series (plot_compare_evokeds) internally to your CI function, unless overwritten to a `ci` param in `ts_args`.

jona-sassenhagen, 2019-05-04:
I can do it too.

mmagnuski, 2019-05-04:
@jona-sassenhagen Sure, If you have a plan, go for it. :)

PR #6270 (2019-07-26): MRG Default ica.plot_properties to a parametric CI
Closes #6268

I am getting approx. half a second speedup for the tests on my machine.

Note I changed the functionality of plot_compare_evokeds a bit: now, as the doctstring already said, it will skip the CI if any condition has fewer than 2 evokeds.

@mmagnuski
closed 2019-07-26 by agramfort