mne-tools/mne-python #6269: Website documentation.html buttons don't work

jona-sassenhagen, 2019-05-04:
On the website, when I am on https://mne-tools.github.io/dev/documentation.html, clicking the buttons (Statistics, Realtime, Connectivity, ...) does not work. The address bar will read, e.g., `https://mne-tools.github.io/dev/documentation.html#collapse_visualization`, but I won't be sent there.

I can click on the buttons from https://mne-tools.github.io/dev/index.html and it will work though.

I tested this on Mac OS on Safari and Firefox.

agramfort, 2019-05-04:
I confirm the issue

@larsoner any idea?

>

larsoner, 2019-05-04:
This was some @choldgraf magic IIRC

I'll try to look soon

drammock, 2019-05-04:
That page on martinos.org/mne/dev does work though. Puzzling.

On May 4, 2019 7:53:31 AM AKDT, Eric Larson <notifications@github.com> wrote:
>This was some @choldgraf magic IIRC
>
>I'll try to look soon

choldgraf, 2019-05-04:
huh, something must have changed on that page. I'm pretty sure it's using the bootstrap collapse/show functionality so maybe something there changed?