mne-tools/mne-python #6271: `raw.plot` for ICA sources scalings are bad

jona-sassenhagen, 2019-05-05:
`ica.get_sources(raw).plot()` yields a largely useless plot, presumably because ICA activities are rescaled by the defaults for the channel type. Would it make sense to allow this to work by default (without the user passing scalings manually)?

agramfort, 2019-05-05:
as sources get returned as MISC channels it's a matter of setting a good
default fo misc channels.

my 2c

>

jona-sassenhagen, 2019-05-06:
Maybe @stfnrpplngr has an idea?

larsoner, 2019-05-06:
Maybe we should always by default auto-scale `misc` channel types in `viz` plotting. By definition we don't have an idea of what the scales should be, so auto-scaling them makes sense.

agramfort, 2019-05-06:
that would be fine with me.

>

mmagnuski, 2019-05-06:
+1 to autoscaling misc, had some issues with these in the past.

PR #6320 (2019-05-16): MRG, ENH: Auto-scale misc channels by default
Closes #6271.

Tested on `master` by adding `ica.get_sources(raw).plot()` on line 90 of `plot_artifacts_correction_ica.py` which looks like:

[image]

And on this PR:

[image]
closed 2019-05-16 by larsoner