mne-tools/mne-python #6278: Volume labels using AAL atlas

ckiefer0, 2019-05-07:
Hi everyone,

I created a [small script](https://gist.github.com/ckiefer0/9b742305db2c7d2d47b8af776a6047f4) to assign volume labels to the voxels in the subject brain. It would be great if I could get some feedback from someone more familiar with MRI/ MNI/ volume source spaces.

The problem is that volume labels are currently not implemented in mne-python, however, the [AAL atlas](http://www.gin.cnrs.fr/en/tools/aal-aal2/) provides an anatomical parcellation of the MNI Collins brain.

My script translates the source space coordinates from the subject brain to MNI space and subsequently assigns labels to the source space coordinates. I implemented three slightly different methods how the labels are assigned in the end.

Generally, the results do not seem to be too bad but depending on the method there are between 500 and 4000 out of 14,629 voxels in the fsaverage brain which do not get a proper label (labelled 'unused' in my script).

agramfort, 2019-05-09:
@ckiefer0 did you consider using nilearn and NiftiLabelsMasker?

https://nilearn.github.io/auto_examples/03_connectivity/plot_signal_extraction.html

larsoner, 2019-05-09:
This looks like the type of functionality we could use. Maybe supporting this sort of labeling scheme / object in our `extract_label_time_course` is the way to go.

ckiefer0, 2019-05-10:
I see how I can use this to extract time series based on the AAL atlas if I transform my source time courses to nifti using `stc.as_volume()`.
However, this does not help me if I just want to know which voxels of my source space belong to which label, e.g., in my case I have created clusters using spatiotemporal clustering and I would like to know which cluster is part of which label.

agramfort, 2019-05-10:
seems a legitimate use case. We could review a PR. Before coding, what
would be the API?
what inputs for what outputs?

>

ckiefer0, 2019-05-15:
I wrote a function to get AAL labels for a subject without all the unnecessary parts like visualization. It is in the gist in aal_utils.py Currently, you can call it like this.
`get_aal_labels(subject, subjects_dir, vsrc, aal_path, min_dist=5)`
Maybe it is something we can build upon. But in the end we should probably try to have the same functionality as with surface source spaces, i.e., label objects etc..

agramfort, 2019-05-15:
can you see now to reuse nilearn here as much as possible?
nilearn offers many atlases (not just AAL) with a consistent API.
It can fetch them too etc.

ckiefer0, 2019-05-29:
I updated the [gist](https://gist.github.com/ckiefer0/9b742305db2c7d2d47b8af776a6047f4). There is a function in `atlas_utils.py` which allows you to get the labels for different atlases if you provide `atlas_data`, `affine` and `value_to_label_dict` which translates the values in `atlas_data` to label names.

The function can be called like this: 
`get_atlas_volume_labels(subject, subjects_dir, vsrc, atlas_data, affine, value_to_label_dict, min_dist=5)`

In `atlas_tests.py` you can try out the different atlases by changing `method` and you can see how to prepare the data before parsing it to `get_atlas_volume_labels()`.

larsoner, 2019-05-29:
>     atlas_data : np.array with 3 dimensions.
>         Array containing the atlas data for each voxel.
>     affine : np.array of shape (4, 4)
>         The affine transformation from indices to coordinates
>         given by nibabel.nifti1.Nifti1Image.affine

Rather than provide these separately it would probably make sense just to take a nibibel SpatialImage as an input, since the simplest form of a SpatialImage is IIRC just a data+affine.

ckiefer0, 2019-05-29:
At fist I was just providing the image to `get_atlas_volume_labels()`. The problem was that the atlas data sometimes has 3 or 4 dimensions and depending on the atlas the data has to be interpreted in a different way (e.g., ICA components or probability of cluster affiliation). To keep the function relatively simple I decided to manipulate the data outside of the function so that I will always have a 3D data array as input. Then the user can decide what he wants to do with the data independent of `get_atlas_volume_labels()`.

Or did you mean it would be better to create a new image object from the manipulated atlas_data plus the affine?

larsoner, 2019-05-29:
> Or did you mean it would be better to create a new image object from the manipulated atlas_data plus the affine?

I would expect the user to get their atlas into `SpatialImage` format (with integer values for each ROI) and pass that to the function.

- in cases where users have to do some manipulations / restrictions / whatever, it only adds one line for them to put it in this format, so it's easy
- even in these cases, they'll probably/often want to save the "adjusted" version to disk
- for cases where the atlas is already in the correct format, users can directly pass Nibabel `img` (or even a filename) not `img.get_data(), img.affine` separately

ckiefer0, 2019-05-29:
OK, I updated the files so that `get_atlas_volume_labels()` takes the argument `atlas_img` which is a nibabel `SpatialImage` object.

larsoner, 2019-05-29:
Great. I have some volumetric data to analyze in the coming days and I'll see how this works with it.

larsoner, 2019-09-19:
> can you see now to reuse nilearn here as much as possible?

@ckiefer0 did you look into this? I looked very briefly at what `nilearn` had at one point and it sounds a lot like what you've been implementing (including support for 3D and 4D atlases).

I have a feeling that the topic of this issue should be solvable with `nilearn` plus the API that @agramfort proposed at the end of #6155

ckiefer0, 2019-09-19:
I haven't spent any time on this recently. What you see in the gist is the current state where I fetch different atlases with nilearn and then use get_atlas_volume_labels() to assign labels to the voxels in the source space.

larsoner, 2019-10-31:
I'm going to close this in favor of #6155 to consolidate future discussions

closed 2019-10-31 by larsoner