mne-tools/mne-python #6279: 3D connectivity plots

fraimondo, 2019-05-07:
#### Describe the problem
After I started working in a lab where @kingjr was, it became more and more often the request to plot his "flaming heads". I was fine, until two days ago.

Just to clarify, the flaming heads are the sensor space connectivity 3d plots (from JR's wSMI paper): 
[image]

#### Describe your solution
So I wrote a couple of _scripts_ (horrible for now) that are able to do something like the image below, using Mayavi, some parts of the `plot_alignment` function, and the [`bezier package.`](https://github.com/dhermes/bezier)

[image]

If you feel like having something like this in MNE-Python, I'm more than happy to work on a PR.

I have a couple of features that I would like to add, including the possibility to do ROI to ROI connectivity (a and b from JR's paper).

This issue was just to open the discussion. I have to ways of plotting, but they depend on my limited knowledge on Mayavi. I have the impression that with some help we can make something better that can be also used to create faster figures.

larsoner, 2019-05-07:
Looks like it could indeed be useful!

@jona-sassenhagen you want to work out the aesthetic considerations / options?

jona-sassenhagen, 2019-05-07:
3D + MEG = not my thing :) But it looks very cool!

agramfort, 2019-05-08:
I would put this for now in mne-sandbox or a separate repo. I don't want to
make
the work of 3D viz refactoring even more challenging

kingjr, 2019-05-08:
Nice! Would be nice to have, I regularly get asked about this and I can
only provide a buggy Matlab code (although I suspect the requests are more
for aesthetic purposes than for scientific ones...).

I m happy to review a PR here or on MNE sandbox, although the main rep will
clearly allow a fancy tweet ;)

> I would put this for now in mne-sandbox or a separate repo. I don't want
to
> make
> the work of 3D viz refactoring even more challenging
https://ci5.googleusercontent.com/proxy/vjTJDvZvCmSQrjVx-Pgy6YYZtNdIh_KXY0aLmmwHQA7tMcKLOfsh1HiEwB2BFKGBgxe9igmFViF8SkOON2kjK0FKcbR3dIZten_UZPYPAUQa-U_oGXXYWH48iotojyNZNQj5ms3_cWLjhB5x6Uuy96_dW35ZLQVjur_P1sGerEiMl8N-2TREKxvMwhZE2IWozDh6dAI-ubu6fr1eThgC10bhKixL12Tpf4rvwqtlXA=s0-d-e1-ft#https://github.com/notifications/beacon/ABFHWDA22IWWXGN5S72BYUDPUKIUXA5CNFSM4HLIDZYKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODU5QHGY.gif>

fraimondo, 2019-05-22:
Finally had some time to make a working example:

https://gist.github.com/fraimondo/8841c53de16aaa63d5953aaed1539f56

Any thoughts on how to proceed? Mostly the bezier curves are the complicated issue. The rest is MNE code.

larsoner, 2019-05-22:
> Any thoughts on how to proceed? Mostly the bezier curves are the complicated issue. The rest is MNE code.

Why not use mpl directly?

https://matplotlib.org/gallery/api/quad_bezier.html

fraimondo, 2019-05-22:
If I’m not mistaken, I can’t do 3D bezier curves with mpl.

larsoner, 2019-05-22:
Ahh right 3D not 2D.

> Any thoughts on how to proceed?

Assuming it's not too much more work (?), a reasonable next step would be to make a WIP PR to make a new viz function for this sort of thing, with an example (or modify an existing example) so we can see what the real-world use case looks like.

If it requires `bezier` to work, we can add it to the list of optional dependencies and put it in `environment.yml` and `requirements.txt`. It appears to be `pip`-installable so hopefully not too bad.

massich, 2019-05-27:
I'm +1 for having this as an example