mne-tools/mne-python #6287: vtki → vista

banesullivan, 2019-05-08:
We recently had to change the name of `vtki` to `vista` per https://github.com/pyvista/vista/issues/208

This had to happen rather quickly and I apologize for any build issues and confusion this might be causing! 

I'll help you all implement the refactoring for this in a PR

larsoner, 2019-05-09:
Fortunately all of our code uses the `vtki` name internally so the switch probably won't be so bad. @GuillaumeFavelier can you give it a shot?

banesullivan, 2019-05-09:
Perfect! I just refactored some of my other projects that use `vista` internally and a simple, project-wide find-and-replace did the trick

Just be sure to limit vista's minimum dependency to version `0.19.0`

banesullivan, 2019-05-10:
Sorry, you all... we're going to rename once more to avoid potential legal issues with the name `vista` not being very unique

We're settling on "PyVista" and will redeploy on PyPI as `pyvista`

https://github.com/pyvista/vista/issues/208

PR #6301 (2019-05-14): MRG, MAINT: Migrate VTKi -> PyVista
Closes #6287.
closed 2019-05-14 by agramfort