mne-tools/mne-python #6295: mark_flats results in annotations that are out of data range

kalenkovich, 2019-05-12:
#### Describe the bug
`mne.preprocessing.mark_flats` appends flat segments to `raw.annotations` that are out of data range.

This probably happens because `mne.preprocessing.mark_flats` adds annotations relative to the first sample while `set_annotations` has changed in `0.17` so that `orig_time` matches `info['meas_date']` and not the first sample.

#### Steps and/or code to reproduce

```Python
import numpy as np
import mne

# Data
sfreq = 1000  # Sampling frequency
times = np.arange(0, 10, 0.001)  # Use 10000 samples (10s)

sin = np.sin(times * 10)  # Multiplied by 10 for shorter cycles
flat = np.zeros(20)

data = np.atleast_2d(np.array(np.r_[sin, flat, sin]))

# Info
ch_types = ['mag']
ch_names = ['sin_flat_sin']

info = mne.create_info(ch_names=ch_names, sfreq=sfreq, ch_types=ch_types)
info['meas_date'] = (1545655654, 71165)

# Raw
raw = mne.io.RawArray(data, info, first_samp=50000)

# Mark flats
mne.preprocessing.flat.mark_flat(raw)

# The annotation is omitted though it should not be
raw.set_annotations(raw.annotations)
```

#### Expected results
Nothing changes when I run `raw.set_annotations(raw.annotations)`

#### Actual results
The segment is omitted due to being out of data range.

```
<stdin>:1: RuntimeWarning: Omitted 1 annotation(s) that were outside data range.
```

#### Additional information
```
Platform:      Windows-7-6.1.7601-SP1
Python:        3.7.3 (default, Apr 24 2019, 15:29:51) [MSC v.1915 64 bit (AMD64)
]
Executable:    D:\Windows_software\Miniconda3\envs\mne_dev_test\python.exe
CPU:           Intel64 Family 6 Model 69 Stepping 1, GenuineIntel: 4 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.18.dev0
numpy:         1.16.3 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.2.1
matplotlib:    Not found

sklearn:       Not found
nibabel:       Not found
mayavi:        Not found
cupy:          Not found
pandas:        Not found
dipy:          Not found
```

agramfort, 2019-05-12:
@kalenkovich thanks for the bug report. Do you want to try to fix it yourself with a pull request?

kalenkovich, 2019-05-13:
@agramfort I'll do it tomorrow.

PR #6310 (2019-05-15): MRG: shifted onsets of annotations created by mark_flats
Onsets were calculated relative to the first sample and not to the `raw.info['meas_date']`.

#### Reference issue
Fixes #6295.

#### What does this implement/fix?
I shifted the onsets by `raw.first_samp / raw.info['sfreq']` to the right so that their timing is relative to  `raw.info['meas_date']`.
closed 2019-05-15 by agramfort