mne-tools/mne-python #6298: BUG, VIZ saving an animated topomap fails

jona-sassenhagen, 2019-05-13:
```
import mne
path = mne.datasets.kiloword.data_path()
fname = path + '/kword_metadata-epo.fif'
evoked = mne.read_epochs(fname).average()
fig, anim = evoked.animate_topomap()
anim.save("topo.gif")
```

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/jona/miniconda3/lib/python3.6/site-packages/matplotlib/animation.py", line 1174, in save
    writer.grab_frame(**savefig_kwargs)
  File "/Users/jona/miniconda3/lib/python3.6/contextlib.py", line 99, in __exit__
    self.gen.throw(type, value, traceback)
  File "/Users/jona/miniconda3/lib/python3.6/site-packages/matplotlib/animation.py", line 232, in saving
    self.finish()
  File "/Users/jona/miniconda3/lib/python3.6/site-packages/matplotlib/animation.py", line 358, in finish
    self.cleanup()
  File "/Users/jona/miniconda3/lib/python3.6/site-packages/matplotlib/animation.py", line 395, in cleanup
    out, err = self._proc.communicate()
  File "/Users/jona/miniconda3/lib/python3.6/subprocess.py", line 863, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
  File "/Users/jona/miniconda3/lib/python3.6/subprocess.py", line 1525, in _communicate
    selector.register(self.stdout, selectors.EVENT_READ)
  File "/Users/jona/miniconda3/lib/python3.6/selectors.py", line 351, in register
    key = super().register(fileobj, events, data)
  File "/Users/jona/miniconda3/lib/python3.6/selectors.py", line 237, in register
    key = SelectorKey(fileobj, self._fileobj_lookup(fileobj), events, data)
  File "/Users/jona/miniconda3/lib/python3.6/selectors.py", line 224, in _fileobj_lookup
    return _fileobj_to_fd(fileobj)
  File "/Users/jona/miniconda3/lib/python3.6/selectors.py", line 39, in _fileobj_to_fd
    "{!r}".format(fileobj)) from None
ValueError: Invalid file object: <_io.BufferedReader name=18>
```

Same in notebook, ipython, REPL.

larsoner, 2019-05-13:
This looks like a `matplotlib` bug, can you see if you can replicate with a simple `mpl` snippet instead?

jona-sassenhagen, 2019-05-13:
You mean manually creating an animation where I loop over time steps for a topomap?

larsoner, 2019-05-13:
No not even a topomap, a script with only `plt` calls (nothing from MNE). In other words, can you get any matplotlib animation to save? Is the problem specific to MNE topomaps, or are all mpl animation saves broken for you?

jona-sassenhagen, 2019-05-13:
Oh no, I can in principle save animations with matplotlib's animation functionality.

larsoner, 2019-05-13:
Okay so we do something with our plotting that does not work for them then. The question is what is it? The traceback you pasted has only `matplotlib` in the traceback so I'm not sure how best to debug.

FWIW this works on my machine (modified to use a smaller time snippet so it does not take forever):
```
import mne
path = mne.datasets.kiloword.data_path()
fname = path + '/kword_metadata-epo.fif'
evoked = mne.read_epochs(fname).average()
evoked.crop(0, 0.1)
fig, anim = evoked.animate_topomap()
anim.save("topo.gif")
```
I get:

[image]

But I get this message:
```
MovieWriter ffmpeg unavailable; trying to use <class 'matplotlib.animation.PillowWriter'> instead.
```
So maybe it's a matplotlib corner case bug where MNE code is used with the `ffmpeg` writer?

jona-sassenhagen, 2019-05-13:
It works for me if I use `writer='imagemagick'` in the save call - and that seems to be the case universally. So not MNE related.

closed 2019-05-13 by jona-sassenhagen