mne-tools/mne-python #6302: MAINT: Release 0.18

larsoner, 2019-05-13:
I am hoping that we can push out a release this week. I know it's a bit short notice, but we are already overdue for our ~6 mo. schedule, and there aren't that many issues/PRs marked for 0.18 anyway. I've gone through and marked issues/PRs that seem doable this week / appropriate for 0.18:

https://github.com/mne-tools/mne-python/milestone/14

Everyone feel free to look at the list of open issues/PRs and let me know if there are others you think should be included. Once there are no remaining issues in the list, I'll push out a release.

GuillaumeFavelier, 2019-05-16:
Can I volunteer for the releasing process described in https://github.com/mne-tools/mne-python/wiki/How-to-make-a-release (@sik promised to help me)

larsoner, 2019-05-16:
That would be great

massich, 2019-05-16:
We'll be releasing tomorrow morning (Paris time). cc: @GuillaumeFavelier

GuillaumeFavelier, 2019-05-17:
Actually I have a pretty busy schedule today so I won't be available much but I cleared my weekend, is it still okay for you?

massich, 2019-05-17:
Sure I've no trouble we'll find a nice coffee shop and spent the Saturday there. RELEASING MNE :D

Is everybody ok with it, or shall we do it today? @agramfort

larsoner, 2019-05-17:
It might be better to do it today

massich, 2019-05-17:
then either you or I can do it

massich, 2019-05-17:
let's see what the CI in master says

larsoner, 2019-05-17:
Yeah once https://travis-ci.org/mne-tools/mne-python/builds/533853924 and https://circleci.com/gh/mne-tools/mne-python/13179 come back happy, feel free to push out 0.18

GuillaumeFavelier, 2019-05-17:
I'm available to work on it now, what do you say @massich ?

larsoner, 2019-05-17:
Just pushed https://github.com/mne-tools/mne-python/commit/3b75010ffdffabf1146f7ca7700e51f93bba3ac6 to fix https://circleci.com/gh/mne-tools/mne-python/13179, now we need to wait 1.5 hours for https://circleci.com/gh/mne-tools/mne-python/13181 to complete

larsoner, 2019-05-17:
https://circleci.com/gh/mne-tools/mne-python/13186 is going, but it's probably better to wait until #6335 builds fully, gets merged, then builds fully on `master`.

@GuillaumeFavelier you might end up getting your wish to work on a Saturday to do the release :)

GuillaumeFavelier, 2019-05-17:
I'm going through the authors list and having troubles with my laptop so tomorrow would be perfect :D

GuillaumeFavelier, 2019-05-18:
@jeythekey we don't have your name to update the mailing for the release. Can you provide us your full name and desired email to ament that? thx.

massich, 2019-05-18:
here's the [mne v0.17 release note](https://github.com/mne-tools/mne-python/issues/5706)

massich, 2019-05-18:
RFC for 0.18 release note is in #6338

massich, 2019-05-18:
We can't release without #6340

massich, 2019-05-19:
I do not understand why stable and dev websites render like this:
[image]

[image]

massich, 2019-05-19:
7ae4626 should fix it. I thought I had pushed that already in `maint/0.18`

massich, 2019-05-19:
now it should be good! @larsoner you can upload martions.

GuillaumeFavelier, 2019-05-19:
This version is live on [PyPi](https://pypi.org/project/mne/0.18.0/#history)!

massich, 2019-05-19:
for completness:

```sh
/tmp
(release-mne-018) ❯ pip search mne      
mne (0.18.0)              - MNE python project for MEG and EEG data analysis.
mne-bids (0.2)            - Experimental code for BIDS using MNE.
mne-hcp (0.1.dev12)       - MNE HCP project for accessing the human connectome MEG data in Python.
mnespikes (0.1a)          - A simple package for representing neural spike data and converting to MNE
okiti-cnt-to-cnt (0.0.3)  - Create mne readable cnt files from OKITI created ones
fieldtrip2mne (0.0.7)     - Convert MEG and EEG brain scan data from FieldTrip toolbox in Matlab to
                            MNE toolbox in python.

/tmp
(release-mne-018) ❯ pip install mne  
Collecting mne
  Downloading https://files.pythonhosted.org/packages/c0/96/a968294f3f21a7e028869bfd8e78f5c5a812232c6ab84091ec2ce578e68b/mne-0.18.0.tar.gz (6.3MB)
     |████████████████████████████████| 6.3MB 2.0MB/s 
Collecting numpy>=1.11.3 (from mne)
  Using cached https://files.pythonhosted.org/packages/bb/76/24e9f32c78e6f6fb26cf2596b428f393bf015b63459468119f282f70a7fd/numpy-1.16.3-cp37-cp37m-manylinux1_x86_64.whl
Collecting scipy>=0.17.1 (from mne)
  Using cached https://files.pythonhosted.org/packages/5d/bd/c0feba81fb60e231cf40fc8a322ed5873c90ef7711795508692b1481a4ae/scipy-1.3.0-cp37-cp37m-manylinux1_x86_64.whl
Building wheels for collected packages: mne
  Building wheel for mne (setup.py) ... done
  Stored in directory: /home/sik/.cache/pip/wheels/49/3b/23/b10a8d761c86821e19511afaba5e266f77c66b19af4c7cffb0
Successfully built mne
Installing collected packages: numpy, scipy, mne
Successfully installed mne-0.18.0 numpy-1.16.3 scipy-1.3.0 

/tmp 34s
(release-mne-018) ❯ python -c "import mne; print(mne.__file__); mne.sys_info()"
/home/sik/miniconda3/envs/release-mne-018/lib/python3.7/site-packages/mne/__init__.py
Platform:      Linux-4.15.0-46-generic-x86_64-with-debian-buster-sid
Python:        3.7.3 | packaged by conda-forge | (default, Mar 27 2019, 23:01:00)  [GCC 7.3.0]
Executable:    /home/sik/miniconda3/envs/release-mne-018/bin/python
CPU:           x86_64: 4 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.18.0
numpy:         1.16.3 {blas=openblas, lapack=openblas}
scipy:         1.3.0
matplotlib:    Not found

sklearn:       Not found
nibabel:       Not found
mayavi:        Not found
cupy:          Not found
pandas:        Not found
dipy:          Not found

```

teonbrooks, 2019-05-19:
> The version **0.18** is live on [PyPi](https://pypi.org/project/mne/0.18.0/#history)!

> We can't release without #6340

I believe we would like to get this #6340 merged in before the release is out. does pypi allow for re-releases or was this a test run?

GuillaumeFavelier, 2019-05-19:
@teonbrooks do you mean #6342 ?

The idea was to release a release candidate to wait for #6342 . But we messed it up. But don't worry, we can do another one quick and it will be the same.

teonbrooks, 2019-05-19:
ah, ok, gotcha! cool, thanks for the clarification :)

massich, 2019-05-21:
Thx everyone !!!

closed 2019-05-21 by massich