mne-tools/mne-python #6304: VIZ: EEG sensor locations wrong in inset plots

drammock, 2019-05-13:
EEG sensor plot insets seem to be broken in some cases. Here are a couple examples. First, from `plot_compare_evokeds`:

[image]

Similar problem with `evoked.plot(spatial_colors=True):

[image]

Compare those to the sensor positions w/r/t the head on `mne.channels.make_eeg_layout(raw.info).plot()` (`raw.plot_sensors()` gives similar result):

[image]

larsoner, 2019-05-16:
I don't see this as a blocker for 0.18, which we want to get out tomorrow if possible. So moving the milestone to 0.19.

massich, 2019-05-17:
Is this related to #5190 ?

drammock, 2019-05-17:
@massich I don't know the layout plotting code well enough to say whether they're related. But these examples should make a good test case for whatever fix comes in for #5190 (or vice versa)

jona-sassenhagen, 2019-05-28:
Isn't the issue simply that layouts are calculated using both MEG and EEG channels, and EEG channels (in this specific montage) don't go as deep?

In which case this might be working as it should be.

drammock, 2019-05-28:
@jona-sassenhagen to me the issue is that *the same set of EEG sensors* nearly fills the head with `raw.plot_sensors()` or `mne.channels.make_eeg_layout(raw.info).plot()` but appear clustered at the crown of the head in the insets in `evoked.plot()` or `plot_compare_evokeds()` calls. IMO they should look the same.

jona-sassenhagen, 2019-05-28:
Not sure how to deal with that. One issue here is that we make no absolute guarantees here. Consider `outlines='skirt'`.

Arguably, the behaviour should match, but maybe the behaviour derived from calculating all layouts simultaneously is the better one.

larsoner, 2019-05-28:
> Arguably, the behaviour should match, but maybe the behaviour derived from calculating all layouts simultaneously is the better one.

@jona-sassenhagen @drammock can you check my proposal here:

https://github.com/mne-tools/mne-python/issues/3987#issuecomment-298920228

I argue that these plots should always show the sensors reasonably aligned to a ~80cm head. Then it doesn't matter what subselection, etc. you use. For MEG you align using `dev_head_t` and for EEG you're already in head coords so should be fine. It should be more true to the data and head alignment, and avoid these rescaling issues altogether.

Also related: #5472, #5471

drammock, 2019-06-04:
adding another test case for #5472.  Would be really nice to be able to do plots like this in the tutorials.

```python
import os
import mne
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)
raw.crop(tmax=60).load_data()
raw.plot_sensors(ch_type='eeg', show_names=True)
midline_ch_names = ['EEG {:03}'.format(n) for n in (2, 12, 30, 48, 58, 60)]
midline = raw.copy().pick_channels(midline_ch_names)
midline.plot_sensors(show_names=True)
```

[image]

[image]

agramfort, 2019-06-04:
I agree we should really fix this :-/

>

larsoner, 2019-06-04:
I started looking at one point and it was a bit of a nightmare. But I'll see if I can look again soon

massich, 2019-06-05:
so did I when I started digitization I expect `plot_sensors` to get simplified with the refactorings.

PR #7066 (2019-12-11): MRG, BUG: Fix M/EEG topomap plotting
Closes #3987
Closes #4880
Closes #5190 
Closes #5471
Closes #5472 
Closes #6304

Todo:
- [x] ensure our montages actually use the same head sizes (done by dig refactoring)
- [x] Make the `plot_sensors` be in physical coordinates
- [x] make all code assume a 0.095 m head radius
- [x] check that all examples in gist are fixed
- [x] deprecate params: `transform` in `montage`, `outlines`, ...
- [x] get tests to pass again
- [x] change to `sphere` rather than `head_radius` for consistency and flexibility
- [x] change units in `setup_volume_source_space` for cons [... cut]
closed 2019-12-11 by agramfort