mne-tools/mne-python #6315: Forward Solution

jgeller112, 2019-05-14:
Hello, 

I am having issues calculating the forward solution. I get the error:  “Cannot use a homogeneous model in EEG calculations.”  What are some reasons for this error?

jgeller112, 2019-05-14:
Related to this issue, has there been a solution to the following error: 

 Surface inner skull is not completely inside surface outer skull

larsoner, 2019-05-15:
>  I get the error: “Cannot use a homogeneous model in EEG calculations.” What are some reasons for this error?

This means you tried to use a 1-layer BEM instead of a 3-layer BEM when doing forward calculations with EEG channels.

> Surface inner skull is not completely inside surface outer skull

This means that your `inner_skull.surf` overlaps with `outer_skull.surf`. Usually this happens when using the watershed bem, and Freesurfer did a bad job of approximating the inner skull surface (which is created just by expanding the `brain.surf`) and/or the `outer_skull.surf` (which is created by shrinking the `outer_skin.surf`. You can visualize this stuff with `freeview`.

For further questions about how to get usable surfaces, the `mne_analysis` listserv is probably a better place so that more people with analysis knowledge and experiences can chime in and learn. GitHub issues are mostly for reporting bugs and/or enhancement requests. I'll leave this open and re-title so we remember to clarify the error message about 1- vs 3-layer BEM.

PR #6318 (2019-05-15): MRG, DOC: Better forward error message
Closes #6315.
closed 2019-05-15 by larsoner