mne-tools/mne-python #6322: BUG: Intro example uses too much memory

larsoner, 2019-05-15:
Running:
```
$ mprof run plot_introduction.py
$ mprof plot
```
I get:

[image]

Examples that use over ~1.5 GB (and 1.0 GB is a safer goal) are prone to die on CircleCI, e.g.:

https://circleci.com/gh/mne-tools/mne-python/13124

@drammock can you find a way to cut down the memory usage? Often cropping to a shorter duration (e.g., 60 sec) is enough. Or using `preload=False` where possible. Also a new note of this loose requirement in the revamped dev docs might help.

drammock, 2019-05-15:
I'll take a look.  I'm already using `del` and avoiding `preload` where I can, but I can probably find ways to deflate it further.

larsoner, 2019-05-15:
For example adding `stop=30` to the `ica.fit` line gets rid of the first big bump:

[image]

The others probably come from loading the raw data, then applying ICA to it. Both can be alleviated by doing `raw.crop(0, 60)` with a suitable message at the beginning that it's done just to save memory. Doing this on line 71 I get the following memory plot (no idea if the plots are actually still reasonable):

[image]

agramfort, 2019-05-15:
don't use maybe sample_audvis_raw.fif but the filtered version so you can decimate? you should also use decim in ICA rather than crop.

larsoner, 2019-05-15:
> you should also use decim in ICA rather than crop.

I'd rather make this a note in the example as an option for users, rather than actually doing it. Adding `raw.crop(0, 60)` makes the entire example runnable from a memory perspective. Without it (even with `decim` or `stop` in `ica.fit`), we will need a number of additional (possibly more complicated) patches elsewhere to avoid excessive memory consumption.

agramfort, 2019-05-15:
try with crop(0, 180) as it should still provide OK results/images. I am
still a bit puzzled why this happens
given that so many examples should do such things.

>

larsoner, 2019-05-15:
> but the filtered version so you can decimate

`sample_audvis_filt-0-40_raw.fif` is actually both filtered *and* decimated already. It does give good memory performance, just swapping in that filename gives:

[image]

> try with crop(0, 180)

Using the standard raw file and doing `raw.crop(0, 180)` gives:

[image]

Probably not good enough, and if we're going to crop at all we might as well get better memory with `.crop(0, 60)`.

> still a bit puzzled why this happens
> given that so many examples should do such things.

We almost always crop, do `ica.fit(..., stop=...)`, or use the filtered+decimated file.

agramfort, 2019-05-15:
it's hard to evaluate as reducing memory is not the end goal. We need
to make sure
that the analysis results still look correct. I cannot judge this with
just these plots.

drammock, 2019-05-15:
I'm rebuilding the docs locally after swapping in the `filt-0-40` file. If the plots still look good I'll open a PR. I don't like the `crop` approach because it makes some of the plots look crappy.

larsoner, 2019-05-15:
> it's hard to evaluate as reducing memory is not the end goal. We need to make sure that the analysis results still look correct. I cannot judge this with just these plots.

Agreed, these plots do not provide a sufficient condition, just a necessary one. So we should only consider `.crop(0, 60)` (probably also `.crop(0, 120)`) or using the 0-40 file.

closed 2019-05-16 by agramfort

jasmainak, 2019-05-16:
late to the party but I think ICA also takes a decim parameter which you can use to reduce the memory footprint

larsoner, 2019-05-16:
>  you should also use decim in ICA rather than crop.

Yes @agramfort suggested this already above :)

PR #6323 (2019-05-16): FIX: use filtered raw to reduce memory in overview tut
closes #6322