mne-tools/mne-python #6325: RFC: two ways of creating fixed lenght events

massich, 2019-05-16:
I just realized that this could be done in different manners.
https://github.com/mne-tools/mne-python/blob/b31e573bd1bb97ab8daf6e813a6f5fa77ba266c4/examples/io/plot_objects_from_arrays.py#L130

I'm not sure if the results are the results are exactly the same but we could imagine something like this:

```diff
- events = mne.make_fixed_length_events(raw, event_id, duration=duration)
+ events, events_id = events_from_annotations(raw.copy().set_annotations(Annotation(0,1e10,'foo')),
+                                             event_id=event_id,
+                                             chunk_duration=duration)
```

Do you think that we should document this somewhere? avoid this duplication? make sure they return the same? or just ignore the whole thing?

(I'm +1 to ignore it, feel free to close)

BTW, the tutorial I pointed on the top says the events have 50% overlap. I failed to find how the overlap is controlled.

larsoner, 2019-05-16:
Unfortunately I think we have to live with there being multiple ways of doing the same thing in MNE. I think the important thing is that we try to make one good, supported way as obvious as possible. For example there is nothing stopping a user from doing:
```
raw_filt = RawArray(mne.filter.filter_data(raw.get_data(), raw.info['sfreq'], None, 40), raw.info)
```
Instead of just:
```
raw_filt = raw.copy().filter(None, 40)
```
But one is clearly more preferable and discoverable than the other.

I think this `fixed_length_events` business is similar in that we have a nice, visible, used-in-examples way to get fixed length events, and then another way you can hack our functions to make them do it another way using `Annotations`.

closed 2019-05-16 by larsoner

massich, 2019-05-16:
Now re-reading my post I see that it seems I want to substitute `fixed_lenght_events` with some hackish way to overwrite the annotations and get events from there. Far from that. It's just that I was not aware of this function until 2 days ago that I got an email from somebody that wanted to epoch without events nor annotations.

I added this `chunk_duration` for creating events for the sleep tutorial and this has never been released. So if we have to revert, now is the time. 

Said that I don't see any way around. The use cases are different. So I guess we have to live with it.

(and we have a record of these similarities in the issue tracker)