mne-tools/mne-python #6330: [mne] I see no events using "mne.read_epochs_eeglab"

SommersKim, 2019-05-17:
[mne] I see no events using "mne.read_epochs_eeglab"

Hello, I've been using eeglab and ERPlab, 
and got raw '.vhdr' files and also preprocessed '.set' files.
Now I need to use mne, but I see empty lists when I load files in either way; "io.read_raw_brainvision" or "io.read_raw_brainvision."
I cannot run "mne.find_events" because there is no epoch info.
But I see it at raw.events like this :
[   51,     0,     1],
       [  301,     0,     2],
       [  551,     0,     3],
How can I find events?
I need your help! Thank you.

Here is the result of my raw.info.

<Info | 17 non-empty fields
    bads : list | 0 items
    ch_names : list | Fp1, Fp2, F3, F4, C3, C4, P3, P4, O1, ...
    chs : list | 32 items (EEG: 32)
    comps : list | 0 items
    custom_ref_applied : bool | False
    dev_head_t : Transform | 3 items
    dig : list | 31 items (31 EEG)
    events : list | 0 items
    highpass : float | 0.0 Hz
    hpi_meas : list | 0 items
    hpi_results : list | 0 items
    lowpass : float | 125.0 Hz
    meas_date : NoneType | unspecified
    nchan : int | 32
    proc_history : list | 0 items
    projs : list | 0 items
    sfreq : float | 250.0 Hz
    acq_pars : NoneType
    acq_stim : NoneType
    ctf_head_t : NoneType
    description : NoneType
    dev_ctf_t : NoneType
    experimenter : NoneType
    file_id : NoneType
    gantry_angle : NoneType
    hpi_subsystem : NoneType
    kit_system_id : NoneType
    line_freq : NoneType
    meas_id : NoneType
    proj_id : NoneType
    proj_name : NoneType
    subject_info : NoneType
    xplotter_layout : NoneType
>

agramfort, 2019-05-17:
find_events in for a raw file not epochs
you used read_epochs_eeglab and not read_raw_eeglab

>

SommersKim, 2019-05-18:
Thank you.. as a student very new to mne-python, no matter how I tried every mne-python code on googling regarding epochs and events, I cannot read my epochs. It's empty.

<Info | 17 non-empty fields
    bads : list | 0 items
    ch_names : list | Fp1, Fp2, F3, F4, C3, C4, P3, P4, O1, ...
    chs : list | 30 items (EEG: 30)
    comps : list | 0 items
    custom_ref_applied : bool | False
    dev_head_t : Transform | 3 items
    dig : list | 29 items (29 EEG)
    events : list | 0 items
    highpass : float | 0.0 Hz
    hpi_meas : list | 0 items
    hpi_results : list | 0 items
    lowpass : float | 125.0 Hz
    meas_date : NoneType | unspecified
    nchan : int | 30
    proc_history : list | 0 items
    projs : list | 0 items
    sfreq : float | 250.0 Hz

Could you let me know how to make disk event file?

raw = mne.io.read_raw_eeglab(my file path to eeg ".set" file)

mne.find_events(raw, stim_channel=None, output='onset', consecutive='increasing', min_duration=0, shortest_event=2, mask=None, uint_cast=False, mask_type='and', initial_event=False, verbose=None)

result : 
array([], shape=(0, 3), dtype=int32)

GuillaumeFavelier, 2019-05-18:
@SommersKim feel free to close the issue if the solution worked.

SommersKim, 2019-05-18:
Sorry, actually my problem is not solved yet. As a student very new to mne-python... I've tried mne.find_events, but it gave me an empty array.
The '.set' file I've loaded is completely processed on eeglab and had no problem with triggers or epochs and was ready to just grand-average.

but when I load it on mne-python, my event array appears empty. 
In this case, should I make an event file? 

mne.find_events(raw, stim_channel=None, output='onset', consecutive='increasing', min_duration=0, shortest_event=2, mask=None, uint_cast=False, mask_type='and', initial_event=False, verbose=None)

array([], shape=(0, 3), dtype=int32)

drammock, 2019-05-18:
@SommersKim please, do not use the GitHub issue tracker for questions about how to *use* the software. The issue tracker is for feature requests and bug reports. See the [getting help](https://mne-tools.github.io/dev/pre_install.html#getting-help) section of the documentation for a link to the MNE email listserv, where usage questions are appropriate and encouraged.

closed 2019-05-18 by drammock