mne-tools/mne-python #6332: [mne] find_events error : arrays used as indices must be of integer (or boolean) type

SommersKim, 2019-05-17:
Hello, I keep getting this error message. Could you please help me on how to find epochs?
I've got these events :
array([[   51,     0,     1],
       [  301,     0,     2],
       [  551,     0,     3]...
but I get this error message...
Thank you!

---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-337-4e899bc791c2> in <module>
----> 1 events = mne.find_events(raw, stim_channel='Pz')

</Users/soankim/mne-python/mne/externals/decorator.py:decorator-gen-130> in find_events(raw, stim_channel, output, consecutive, min_duration, shortest_event, mask, uint_cast, mask_type, initial_event, verbose)

~/mne-python/mne/utils/_logging.py in wrapper(*args, **kwargs)
     87             with use_log_level(verbose_level):
     88                 return function(*args, **kwargs)
---> 89         return function(*args, **kwargs)
     90     return FunctionMaker.create(
     91         function, 'return decfunc(%(signature)s)',

~/mne-python/mne/event.py in find_events(raw, stim_channel, output, consecutive, min_duration, shortest_event, mask, uint_cast, mask_type, initial_event, verbose)
    684     if len(picks) == 0:
    685         raise ValueError('No stim channel found to extract event triggers.')
--> 686     data, _ = raw[picks, :]
    687 
    688     events_list = []

~/mne-python/mne/utils/mixin.py in __getitem__(self, item)
    129                .. versionadded:: 0.16
    130         """
--> 131         return self._getitem(item)
    132 
    133     def _getitem(self, item, reason='IGNORED', copy=True, drop_event_id=True,

~/mne-python/mne/utils/mixin.py in _getitem(***failed resolving arguments***)
    179         has_selection = hasattr(inst, 'selection')
    180         if has_selection:
--> 181             key_selection = inst.selection[select]
    182             if reason is not None:
    183                 for k in np.setdiff1d(inst.selection, key_selection):

IndexError: arrays used as indices must be of integer (or boolean) type

agramfort, 2019-05-17:
you don't have a stim channel in your raw file. Do you see annotations?
raw.annotations ?

>

SommersKim, 2019-05-17:
No, because it's already epoched on eeglab. I get this error with raw.annotations :

AttributeError: 'EpochsEEGLAB' object has no attribute 'annotations'

agramfort, 2019-05-17:
events are to make epochs. If you have epochs what are you trying to do?

>

SommersKim, 2019-05-17:
You're right. Thank you. Actually, I wanted to run this code, but I think it's not the problem of epochs... 

epochs = mne.read_epochs(path)
(https://martinos.org/mne/stable/auto_examples/stats/plot_sensor_regression.html)

AttributeError: 'EpochsEEGLAB' object has no attribute 'endswith'

But it shows the same error message when I run "mne.io.read_raw_eeglab" as well.
Should I convert all my ".set"files into ".fif" files?
I don't know what the problem is..
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-353-a61e39ab83ed> in <module>
----> 1 epochs = mne.read_epochs(raw)

</Users/soankim/mne-python/mne/externals/decorator.py:decorator-gen-207> in read_epochs(fname, proj, preload, verbose)

~/mne-python/mne/utils/_logging.py in wrapper(*args, **kwargs)
     87             with use_log_level(verbose_level):
     88                 return function(*args, **kwargs)
---> 89         return function(*args, **kwargs)
     90     return FunctionMaker.create(
     91         function, 'return decfunc(%(signature)s)',

~/mne-python/mne/epochs.py in read_epochs(fname, proj, preload, verbose)
   2339         The epochs
   2340     """
-> 2341     return EpochsFIF(fname, proj, preload, verbose)
   2342 
   2343 

</Users/soankim/mne-python/mne/externals/decorator.py:decorator-gen-208> in __init__(self, fname, proj, preload, verbose)

~/mne-python/mne/utils/_logging.py in wrapper(*args, **kwargs)
     87             with use_log_level(verbose_level):
     88                 return function(*args, **kwargs)
---> 89         return function(*args, **kwargs)
     90     return FunctionMaker.create(
     91         function, 'return decfunc(%(signature)s)',

~/mne-python/mne/epochs.py in __init__(self, fname, proj, preload, verbose)
   2393                  verbose=None):  # noqa: D102
   2394         check_fname(fname, 'epochs', ('-epo.fif', '-epo.fif.gz',
-> 2395                                       '_epo.fif', '_epo.fif.gz'))
   2396         fnames = [fname]
   2397         ep_list = list()

~/mne-python/mne/utils/check.py in check_fname(fname, filetype, endings, endings_err)
     46                       % (fname, filetype, print_endings))
     47     print_endings = ' or '.join([', '.join(endings[:-1]), endings[-1]])
---> 48     if not fname.endswith(endings):
     49         warn('This filename (%s) does not conform to MNE naming conventions. '
     50              'All %s files should end with %s'

AttributeError: 'EpochsEEGLAB' object has no attribute 'endswith'

agramfort, 2019-05-17:
it's hard for us to debug without the .set file and seeing the full code

>

SommersKim, 2019-05-18:
I've uploaded my raw BrainProduct eegfiles('001.vhdr', '001.vmrk') and also pre-processed & artifac rejected ".set"files("001.set", "001.fdt").[https://github.com/SommersKim/EEG-ERP](url)

After I loaded my files with "mne.io.read_raw_brainvision" for my raw BrainProduct eegfiles,  or "mne.io.read_epochs_eeglab" for my ".set"files, 

I cannot run "epochs = mne.read_epochs(raw)".

Thank you...!!

agramfort, 2019-05-18:
@SommersKim please read the documentation.

how do you expect `epochs = mne.read_epochs(raw)` to work? read_epochs reads from disk epochs. Why do you pass a raw object???

I am closing as I don't think it's a bug in the code.

closed 2019-05-18 by agramfort